pythops/jetson-image #1: create-rootfs.sh  didn't install uninstalled package

dlarue, 2019-11-28:
debootstrap wasn't installed on my Kubuntu 18.04.3 system and the script would end at the uninstalled package. I put a printf statement in the script to show the package which it ended on and checked that it wasn't installed. After installing it manually the script continued as all other packages were installed.

if [ $? -ne 0 ]; then
                apt install -y $pkg > /dev/null
        fi

pythops, 2019-11-28:
That's weird cause the script does check if debootstrap is installed
```
for pkg in qemu-user-static debootstrap coreutils parted wget gdisk e2fsprogs; do
	dpkg -l | grep $pkg > /dev/null
	if [ $? -ne 0 ]; then
		apt install -y $pkg > /dev/null
	fi
done
```

What is the output of this command when your run it in your system ?
```
$ dpkg -l | grep debootstrap
```

dlarue, 2019-11-28:
it would show bootstrap wasn't installed.
I eventually just added: apt install -y qemu-user-static debootstrap coreutils parted wget gdisk e2fsprogs
since apt will handle if any are or aren't installed.

dlarue, 2019-11-29:
To test it, just put an exit declare after the "# Install prerequisites packages" section and then just change the name of one of the packages so it's not found.

I put some print statements in and for some reason it just stops at the "dpkg -l | grep $pkg..." statement.

pythops, 2019-11-29:
I will remove the for loop and put `apt install -y qemu-user-static debootstrap coreutils parted wget gdisk e2fsprogs` instead

pythops, 2019-11-29:
Done

closed 2019-11-29 by pythops