pythops/jetson-image #2: Running ansible on jetson.yaml fails missing package python3-apt

dlarue, 2019-11-28:
Running from bootable Lubuntu 18.04.3 LiveCD image:

sudo -E $(which ansible-playbook) jetson.yaml

PLAY [Customize Jetson rootfs] ************************************************************************************************************************************************************

TASK [Gathering Facts] ********************************************************************************************************************************************************************
ok: [jetson]

TASK [jetson : Update sources.list] *******************************************************************************************************************************************************
changed: [jetson]

TASK [jetson : Upgrade packages] **********************************************************************************************************************************************************
[WARNING]: Updating cache and auto-installing missing dependency: python3-apt

fatal: [jetson]: FAILED! => {"changed": false, "cmd": "apt-get install --no-install-recommends python3-apt -y -q", "msg": "E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/a/apt/libapt-inst2.0_1.6.1_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'\nE: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python-apt-common_1.6.0_all.deb  Temporary failure resolving 'ports.ubuntu.com'\nE: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python3-apt_1.6.0_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?", "rc": 100, "stderr": "E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/a/apt/libapt-inst2.0_1.6.1_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'\nE: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python-apt-common_1.6.0_all.deb  Temporary failure resolving 'ports.ubuntu.com'\nE: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python3-apt_1.6.0_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "stderr_lines": ["E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/a/apt/libapt-inst2.0_1.6.1_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'", "E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python-apt-common_1.6.0_all.deb  Temporary failure resolving 'ports.ubuntu.com'", "E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python3-apt_1.6.0_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'", "E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following additional packages will be installed:\n  libapt-inst2.0 python-apt-common\nSuggested packages:\n  python3-apt-dbg python-apt-doc\nRecommended packages:\n  lsb-release iso-codes\nThe following NEW packages will be installed:\n  libapt-inst2.0 python-apt-common python3-apt\n0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.\nNeed to get 218 kB of archives.\nAfter this operation, 1423 kB of additional disk space will be used.\nErr:1 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 libapt-inst2.0 arm64 1.6.1\n  Temporary failure resolving 'ports.ubuntu.com'\nErr:2 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-apt-common all 1.6.0\n  Temporary failure resolving 'ports.ubuntu.com'\nErr:3 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python3-apt arm64 1.6.0\n  Temporary failure resolving 'ports.ubuntu.com'\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following additional packages will be installed:", "  libapt-inst2.0 python-apt-common", "Suggested packages:", "  python3-apt-dbg python-apt-doc", "Recommended packages:", "  lsb-release iso-codes", "The following NEW packages will be installed:", "  libapt-inst2.0 python-apt-common python3-apt", "0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.", "Need to get 218 kB of archives.", "After this operation, 1423 kB of additional disk space will be used.", "Err:1 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 libapt-inst2.0 arm64 1.6.1", "  Temporary failure resolving 'ports.ubuntu.com'", "Err:2 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-apt-common all 1.6.0", "  Temporary failure resolving 'ports.ubuntu.com'", "Err:3 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python3-apt arm64 1.6.0", "  Temporary failure resolving 'ports.ubuntu.com'"]}

PLAY RECAP ********************************************************************************************************************************************************************************
jetson                     : ok=2    changed=1    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0

pythops, 2019-11-29:
Check your network, you seem to have issues with DNS.

dlarue, 2019-11-29:
it must be something in the chroot since it was one of the first things I verified when I saw the error.  All I know is that on a clean system, the scripts didn't work.  Grab Lubuntu 18.04.3 and give it a try as a bootable liveCD environment.  You could even run it inside a virtual machine as a live system to test. I could not get it to work.

viteo, 2020-03-11:
the very same issue here. any thoughts?

TWhidden, 2020-04-19:
Not sure if this will help someone else... but this helped me.  I added to the file `/ansible/roles/jetson/tasks/main.yaml` the following lines at the top:

```
- name: setup dns
  template:
    src: etc/resolv.conf
    dest: /etc/resolv.conf
    owner: root
    group: root
    mode: 0644
```
and also created that matching file in 
`/ansible/roles/jetson/templates/etc/resolv.conf`

with `nameserver 8.8.8.8`

that got me past the apt-update issues.  I am building this on a Nano, which can resolve everything on the host, but for whatever reason, fails resolving when running ansible. Anyhow, hope this helps someone.

hosaka, 2020-07-15:
@pythops This is the problem with chroot. The resolv.conf file should be generated by the resolvconf service.
Perhaps worth mentioning it, or at least re-open this issue so it helps others to find the solution.

linux-devil, 2020-08-13:
@TWhidden Thank you , you saved me hours

commit 0964a02e (2021-02-10): Fix dns issue; resolves #2
closed 2021-02-10 by pythops