pythops/jetson-image #5: Can not create image

lhlong, 2020-04-15:
Hello,

Thank for your tutorial at https://pythops.com/post/create-your-own-image-for-jetson-nano-board
I follow step by step on your tutorial, create a root is okay. 
But, I got the error while trying to create the image with this command `sudo -E ./create-image.sh`
Jetson nano 32.3.2
Anyone had the same issue?

> /home/lhjet/data/build/Linux_for_Tegra/flash.sh: line 575: ./tegrarcm: cannot execute binary file: Exec format error
> Board ID(3448) version(200) 
> copying bctfile(/home/lhjet/data/build/Linux_for_Tegra/bootloader/t210ref/BCT/P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg)... done.
> copying bootloader(/home/lhjet/data/build/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
> copying initrd(/home/lhjet/data/build/Linux_for_Tegra/bootloader/l4t_initrd.img)... done.
> 	populating kernel to rootfs... done.
> 	populating initrd to rootfs... done.
> 	populating /home/lhjet/data/build/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0000-p3449-0000-a02.dtb to rootfs... done.
> Making Boot image... failed.

[image]

pythops, 2020-04-15:
Did you update the BSP in `create-image.sh` script ?
```
BSP=https://developer.nvidia.com/embedded/r32-2-3_Release_v1.0/t210ref_release_aarch64/Tegra210_Linux_R32.2.3_aarch64.tbz2
```

lhlong, 2020-04-16:
> Did you update the BSP in `create-image.sh` script ?
> 
> ```
> BSP=https://developer.nvidia.com/embedded/r32-2-3_Release_v1.0/t210ref_release_aarch64/Tegra210_Linux_R32.2.3_aarch64.tbz2
> ```

No, I didn't. 
I checked your script. It had the same version [r32.2.3](https://developer.nvidia.com/jetson-nano-sd-card-image-r3223)

pythops, 2020-04-16:
which OS do you use ?

closed 2020-04-18 by lhlong

TWhidden, 2020-04-19:
@pythops  

Same issue as the poster.

Can you clarify how we should be building this?  I tried to build this under the WSL platform (Ubuntu 18.04), then under a HyperV Ubuntu 18.04 (x64 also) image, but both fail - (bad format exceptions).

I was able to get past the bad format exception by building this on the Nano (arm64 obviously)... but during the `./createimage.sh` process it fails on another `exec format error`

```
Build the image ...
Download L4T...       [OK]
Extract L4T...        [OK]
Create image...       ********************************************
     Jetson-Nano SD Image Creation Tool
********************************************
create-jetson-nano-sd-card-image.sh - creating jetson.img of 2010M...
0+0 records in
0+0 records out
0 bytes copied, 0.000284226 s, 0.0 kB/s
create-jetson-nano-sd-card-image.sh - creating signed images
###############################################################################
# L4T BSP Information:
# R32 , REVISION: 2.3
###############################################################################
/root/jetson_build/Linux_for_Tegra/flash.sh: line 575: ./tegrarcm: cannot execute binary file: Exec format error
Board ID(3448) version(200)
creating dummy /root/jetson_build/Linux_for_Tegra/bootloader/badpage.bin
1+0 records in
1+0 records out
4096 bytes (4.1 kB, 4.0 KiB) copied, 0.000659496 s, 6.2 MB/s
copying bctfile(/root/jetson_build/Linux_for_Tegra/bootloader/t210ref/BCT/P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg)... done.
copying bootloader(/root/jetson_build/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
copying initrd(/root/jetson_build/Linux_for_Tegra/bootloader/l4t_initrd.img)... done.
        populating kernel to rootfs... done.
        populating initrd to rootfs... done.
        populating /root/jetson_build/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0000-p3449-0000-a02.dtb to rootfs... done.
Making Boot image... failed.
```

Can you post your recommendations for building the image on your blog page? At least then we know we am building using the correct cpu platform / OS.   

This is a very useful page you have created - The bloat in the NVidia image is not suitable for a production system at all. Plus adding your ansible components was very helpful.  Many props for this how-to.

Thanks!

TWhidden, 2020-04-19:
@lhlong  - Can you post what you did to get past this issue?  I'm at the same spot.  Thanks.

TWhidden, 2020-04-20:
Just a follow up - looks as if I was able to create this on an Ubuntu 18.04 server / x64 system (not virtual).  

Does not appear to work via Windows WSL: 
`Run debootstrap first stage...  mknod: /mnt/c/Devel/badhare-jetsonimage_rootfs/test-dev-null: Operation not permitted`

And a visualized Ubuntu 18.04 appears to have trouble running the qemu parts preventing it from being successful.

Just a followup in case someone run into this on their own system.

bricknano, 2020-12-18:
Hello, I'm having  the same problem while creating image, i'm using a jetson nano for the process 

`Build the image ...
Download L4T...       [OK]
Extract L4T...        [OK]
Create image...       ********************************************
     Jetson-Nano SD Image Creation Tool     
********************************************
create-jetson-nano-sd-card-image.sh - creating jetson.img of 2028M...
0+0 records in
0+0 records out
0 bytes copied, 0.000369074 s, 0.0 kB/s
create-jetson-nano-sd-card-image.sh - creating signed images
###############################################################################
# L4T BSP Information:
# R32 , REVISION: 2.3
###############################################################################
/home/brick/jetson-headless-image/image-build/Linux_for_Tegra/flash.sh: line 575: ./tegrarcm: cannot execute binary file: Exec format error
Board ID(3448) version(200) 
creating dummy /home/brick/jetson-headless-image/image-build/Linux_for_Tegra/bootloader/badpage.bin
1+0 records in
1+0 records out
4096 bytes (4.1 kB, 4.0 KiB) copied, 0.000408866 s, 10.0 MB/s
copying bctfile(/home/brick/jetson-headless-image/image-build/Linux_for_Tegra/bootloader/t210ref/BCT/P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg)... done.
copying bootloader(/home/brick/jetson-headless-image/image-build/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
copying initrd(/home/brick/jetson-headless-image/image-build/Linux_for_Tegra/bootloader/l4t_initrd.img)... done.
	populating kernel to rootfs... done.
	populating initrd to rootfs... done.
	populating /home/brick/jetson-headless-image/image-build/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0000-p3449-0000-a02.dtb to rootfs... done.
Making Boot image... failed.
`

baxks, 2020-12-20:
> Hello, I'm having the same problem while creating image, i'm using a jetson nano for the process
>
> `Build the image ...
> Download L4T... [OK]
> Extract L4T... [OK]
> Create image... [...]

I had this problem, managed to get around it by doing the first part (create-rootfs and ansible on the jetson nano) 
then copying the rootfs to an ubuntu 20 machine (not VM) and do the second part (create-image)