pythops/jetson-image #6: debootstap stage 2 failed

drSchaps, 2020-06-05:
Hi,
i had a problem with creating image for jetson nano. On freshly installed Ubuntu 18.04 with all updates.  
```sudo -E ./create-rootfs.sh
Install the dependencies...  [OK]
Create rootfs directory...    [OK]
Download the base image...   [OK]
Run debootstrap first stage...  [OK]
Run debootstrap second stage... chroot: failed to run command ‘/bin/bash’: Exec format error
 ```
After some investigation i found solution here: [https://github.com/multiarch/qemu-user-static/issues/49#issuecomment-513911251](url)

Before chrooting you must register qemu-aarch64 with this command:
```
echo ":qemu-aarch64:M::\x7fELF\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xb7\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-aarch64-static:" > /proc/sys/fs/binfmt_misc/register
```
and unregister after chroot with this command:
```
echo -1 > /proc/sys/fs/binfmt_misc/qemu-aarch64 
```
PS: Thank you for your work :)

pythops, 2020-06-16:
Never had that issue before and it works fine for me on both ubuntu 18.04 & ubuntu 20.04

pythops, 2020-06-18:
I'm closing this issue as I can not reproduce this issue.

closed 2020-06-18 by pythops

echosalik, 2021-03-28:
@drSchaps if you are using wsl2 then this does come up but doesn't on a Ubuntu VM.
I guess WSL2 doesnt work well with qemu and emulations.