pythops/jetson-image #9: Flashing doesn't extend fs.

hkarrson, 2020-08-11:
Hey!

I get this error when flashing... 
Extend the fs... e2fsck: No such file or directory while trying to open /dev/sdcp1

I'm not sure if this is related, but when I boot the nano with any of the img's I've created, it says
[FAILED] failed to start Resize SD Card root partition and filesystem.
See 'systemctl status nvresizefs.service' for details.
When I run that command, I get the following output...
nvresizefs.service - Resize SD Card root partition and filesystem
 Loaded: loaded (/etc/systemd/system/nvresizefs.service; enabled; vendor preset: enabled)
 Active: failed (Result: exit-code) since Sun 2018-01-28 15:58:18 UTC; 15min ago
Process: 3225 ExecStart=etc/systemd/nvresizefs.sh (code=exited, status=127)
Main PID: 3225 (code=exited, status=127)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

Thanks!

pythops, 2020-08-11:
>  Extend the fs... e2fsck: No such file or directory while trying to open /dev/sdcp1

can you post here the command you run when you try to flash the sd card and also the output of `lsblk` command

hkarrson, 2020-08-13:
Thank you for your quick response! The command I used is sudo ./flash-image.sh /path/to/build_dir/Linux_for_Tegra/jetson.img /dev/sdc - I literally made /path/to/build_dir the build dir. 

Here's lsblk:

karrson@karrson-linux-dev:~/CustomOS/jetson-nano-image$ lsblk
NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop1     7:1    0    55M  1 loop /snap/core18/1880
sdb       8:16   0 931.5G  0 disk 
loop4     7:4    0  94.5M  1 loop /snap/github-desktop/63
sr0      11:0    1   3.9G  0 rom  /media/karrson/reFX NEXUS 2
loop2     7:2    0  55.3M  1 loop /snap/core18/1885
loop0     7:0    0 131.1M  1 loop /snap/github-desktop/x1
sdc       8:32   1  59.5G  0 disk 
├─sdc2    8:34   1   128K  0 part 
├─sdc13   8:45   1    80K  0 part 
├─sdc11   8:43   1   768K  0 part 
├─sdc9    8:41   1   448K  0 part 
├─sdc7    8:39   1   384K  0 part 
├─sdc5    8:37   1    64K  0 part 
├─sdc3    8:35   1   448K  0 part 
├─sdc14   8:46   1   128K  0 part 
├─sdc1    8:33   1  13.2G  0 part /media/karrson/7266b927-cc74-4440-9d2a-0881d81
├─sdc12   8:44   1    64K  0 part 
├─sdc10   8:42   1   448K  0 part 
├─sdc8    8:40   1    64K  0 part 
├─sdc6    8:38   1   192K  0 part 
└─sdc4    8:36   1   576K  0 part 
sda       8:0    0 931.5G  0 disk 
├─sda2    8:2    0     1K  0 part 
├─sda5    8:5    0   976M  0 part [SWAP]
└─sda1    8:1    0 930.6G  0 part /
loop5     7:5    0  96.5M  1 loop /snap/core/9436
loop3     7:3    0    97M  1 loop /snap/core/9665

**But** I actually found a solution! 
In flash_image.sh, there was a part of the file, I forget where, but it said p1 and I just replaced that with 1 and it found the device.
Thanks! :)

closed 2020-08-13 by hkarrson