pythops/jetson-image #11: Trying to create image using latest BSP

linux-devil, 2020-08-13:
I used BSP and root fs from following webiste  : https://developer.nvidia.com/embedded/linux-tegra

rootfs : https://developer.nvidia.com/embedded/L4T/r32_Release_v4.3/t210ref_release_aarch64/Tegra_Linux_Sample-Root-Filesystem_R32.4.3_aarch64.tbz2

bsp : https://developer.nvidia.com/embedded/L4T/r32_Release_v4.3/t210ref_release_aarch64/Tegra210_Linux_R32.4.3_aarch64.tbz2

However I get error when I try to apply binaries in createimage.sh 

Error : 
```
dpkg: dependency problems prevent configuration of python-jetson-gpio:
 python-jetson-gpio depends on python:any (<< 2.8).
 python-jetson-gpio depends on python:any (>= 2.7.5-5~).

dpkg: error processing package python-jetson-gpio (--install):
 dependency problems - leaving unconfigured
Setting up nvidia-l4t-apt-source (32.4.3-20200625213809) ...
Pre-installing... skip changing source list.
dpkg: dependency problems prevent configuration of nvidia-l4t-camera:
 nvidia-l4t-camera depends on libgles2; however:
  Package libgles2 is not installed.

dpkg: error processing package nvidia-l4t-camera (--install):
 dependency problems - leaving unconfigured
Setting up nvidia-l4t-configs (32.4.3-20200625213809) ...
Setting up nvidia-l4t-firmware (32.4.3-20200625213809) ...
dpkg: dependency problems prevent configuration of nvidia-l4t-graphics-demos:
 nvidia-l4t-graphics-demos depends on libgles2; however:
  Package libgles2 is not installed.
 nvidia-l4t-graphics-demos depends on libwayland-egl1; however:
  Package libwayland-egl1 is not installed.
 nvidia-l4t-graphics-demos depends on libxkbcommon0; however:
  Package libxkbcommon0 is not installed.

dpkg: error processing package nvidia-l4t-graphics-demos (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of nvidia-l4t-gstreamer:
 nvidia-l4t-gstreamer depends on libasound2; however:
  Package libasound2 is not installed.
 nvidia-l4t-gstreamer depends on libgles2; however:
  Package libgles2 is not installed.
 nvidia-l4t-gstreamer depends on libgstreamer1.0-0; however:
  Package libgstreamer1.0-0 is not installed.
 nvidia-l4t-gstreamer depends on libgstreamer-plugins-bad1.0-0; however:
  Package libgstreamer-plugins-bad1.0-0 is not installed.
 nvidia-l4t-gstreamer depends on libgstreamer-plugins-base1.0-0; however:
  Package libgstreamer-plugins-base1.0-0 is not installed.
 nvidia-l4t-gstreamer depends on libwayland-egl1; however:
  Package libwayland-egl1 is not installed.
 nvidia-l4t-gstreamer depends on nvidia-l4t-camera (= 32.4.3-20200625213809); however:
  Package nvidia-l4t-camera is not configured yet.

dpkg: error processing package nvidia-l4t-gstreamer (--install):
 dependency problems - leaving unconfigured
Setting up nvidia-l4t-init (32.4.3-20200625213809) ...
dpkg: dependency problems prevent configuration of nvidia-l4t-jetson-io:
 nvidia-l4t-jetson-io depends on device-tree-compiler; however:
  Package device-tree-compiler is not installed.

dpkg: error processing package nvidia-l4t-jetson-io (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of nvidia-l4t-multimedia:
 nvidia-l4t-multimedia depends on libasound2; however:
  Package libasound2 is not installed.
 nvidia-l4t-multimedia depends on libcairo2; however:
  Package libcairo2 is not installed.
 nvidia-l4t-multimedia depends on libdatrie1; however:
  Package libdatrie1 is not installed.
 nvidia-l4t-multimedia depends on libfontconfig1; however:
  Package libfontconfig1 is not installed.
 nvidia-l4t-multimedia depends on libgles2; however:
  Package libgles2 is not installed.
 nvidia-l4t-multimedia depends on libgstreamer1.0-0; however:
  Package libgstreamer1.0-0 is not installed.
 nvidia-l4t-multimedia depends on libgstreamer-plugins-bad1.0-0; however:
  Package libgstreamer-plugins-bad1.0-0 is not installed.
 nvidia-l4t-multimedia depends on libgstreamer-plugins-base1.0-0; however:
  Package libgstreamer-plugins-base1.0-0 is not installed.
 nvidia-l4t-multimedia depends on libharfbuzz0b; however:
  Package libharfbuzz0b is not installed.
 nvidia-l4t-multimedia depends on libpango-1.0-0; however:
  Package libpango-1.0-0 is not installed.
 nvidia-l4t-multimedia depends on libpangocairo-1.0-0; however:
  Package libpangocairo-1.0-0 is not installed.
 nvidia-l4t-multimedia depends on libpangoft2-1.0-0; however:
  Package libpangoft2-1.0-0 is not installed.
 nvidia-l4t-multimedia depends on libpixman-1-0; however:
  Package libpixman-1-0 is not installed.
 nvidia-l4t-multimedia depends on libxrender1; however:
  Package libxrender1 is not installed.

dpkg: error processing package nvidia-l4t-multimedia (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of nvidia-l4t-multimedia-utils:
 nvidia-l4t-multimedia-utils depends on libcairo2; however:
  Package libcairo2 is not installed.
 nvidia-l4t-multimedia-utils depends on libdatrie1; however:
  Package libdatrie1 is not installed.
 nvidia-l4t-multimedia-utils depends on libfontconfig1; however:
  Package libfontconfig1 is not installed.
 nvidia-l4t-multimedia-utils depends on libgles2; however:
  Package libgles2 is not installed.
 nvidia-l4t-multimedia-utils depends on libharfbuzz0b; however:
  Package libharfbuzz0b is not installed.
 nvidia-l4t-multimedia-utils depends on libpangoft2-1.0-0; however:
  Package libpangoft2-1.0-0 is not installed.
 nvidia-l4t-multimedia-utils depends on libpixman-1-0; however:
  Package libpixman-1-0 is not installed.
 nvidia-l4t-multimedia-utils depends on libxrender1; however:
  Package libxrender1 is not installed.

dpkg: error processing package nvidia-l4t-multimedia-utils (--install):
 dependency problems - leaving unconfigured
Setting up nvidia-l4t-oem-config (32.4.3-20200625213809) ...
Setting up nvidia-l4t-tools (32.4.3-20200625213809) ...
Setting up nvidia-l4t-wayland (32.4.3-20200625213809) ...
dpkg: dependency problems prevent configuration of nvidia-l4t-weston:
 nvidia-l4t-weston depends on libcairo2; however:
  Package libcairo2 is not installed.
 nvidia-l4t-weston depends on libevdev2; however:
  Package libevdev2 is not installed.
 nvidia-l4t-weston depends on libgles2; however:
  Package libgles2 is not installed.
 nvidia-l4t-weston depends on libinput10; however:
  Package libinput10 is not installed.
 nvidia-l4t-weston depends on libjpeg-turbo8; however:
  Package libjpeg-turbo8 is not installed.
 nvidia-l4t-weston depends on libpixman-1-0; however:
  Package libpixman-1-0 is not installed.
 nvidia-l4t-weston depends on libpng16-16; however:
  Package libpng16-16 is not installed.
 nvidia-l4t-weston depends on libunwind8; however:
  Package libunwind8 is not installed.
 nvidia-l4t-weston depends on libwayland-cursor0; however:
  Package libwayland-cursor0 is not installed.
 nvidia-l4t-weston depends on libwayland-egl1; however:
  Package libwayland-egl1 is not installed.
 nvidia-l4t-weston depends on libxkbcommon0; however:
  Package libxkbcommon0 is not installed.

dpkg: error processing package nvidia-l4t-weston (--install):
 dependency problems - leaving unconfigured
Setting up nvidia-l4t-x11 (32.4.3-20200625213809) ...
Setting up nvidia-l4t-xusb-firmware (32.4.3-20200625213809) ...
Pre-installing xusb firmware package, skip flashing
Setting up nvidia-l4t-kernel (4.9.140-tegra-32.4.3-20200625213809) ...
dpkg: dependency problems prevent configuration of nvidia-l4t-kernel-dtbs:
 nvidia-l4t-kernel-dtbs depends on device-tree-compiler; however:
  Package device-tree-compiler is not installed.

dpkg: error processing package nvidia-l4t-kernel-dtbs (--install):
 dependency problems - leaving unconfigured
Setting up nvidia-l4t-kernel-headers (4.9.140-tegra-32.4.3-20200625213809) ...
Unpacking kernel headers...
Setting up nvidia-l4t-bootloader (32.4.3-20200625213809) ...
Pre-installing bootloader package, skip flashing
Setting up nvidia-l4t-3d-core (32.4.3-20200625213809) ...
Setting up nvidia-l4t-cuda (32.4.3-20200625213809) ...
Setting up nvidia-l4t-initrd (32.4.3-20200625213809) ...
Processing triggers for nvidia-l4t-kernel (4.9.140-tegra-32.4.3-20200625213809) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Errors were encountered while processing:
 python-jetson-gpio
 nvidia-l4t-camera
 nvidia-l4t-graphics-demos
 nvidia-l4t-gstreamer
 nvidia-l4t-jetson-io
 nvidia-l4t-multimedia
 nvidia-l4t-multimedia-utils
 nvidia-l4t-weston
 nvidia-l4t-kernel-dtbs

```

pythops, 2020-08-13:
The current scripts are not compatible with the latest BSP. I  may update them soon, meanwhile any PR are welcome.

linux-devil, 2020-08-14:
Looking forward to it

Dandjinh, 2020-11-22:
1. run **create-image.sh** until it exits with errors
2. **chroot** to the $JETSON_BUILD_DIR/Linux_for_Tegra/rootfs and run **apt-get --fix-broken install**
3. run **$JETSON_BUILD_DIR/Linux_for_Tegra//tools/jetson-disk-image-creator.sh nano-image.img -b jetson-nano -r 300**
4. finally, get the image

pythops, 2021-02-03:
Done

closed 2021-02-03 by pythops