pythops/jetson-image #20: jetson-nano 2Gb bootloader error

natdop, 2021-02-10:
Hello,

I have several problems with the image when I boot, first I have a DNS resolving problem, I could temporary solve this by adding `nameserver 8.8.8 to /etc/resolv.conf`

The other problem is with apt, I can't install packages or upgrade: 
```
Errors were encountered while processing:
nvidia-l4t-bootloader
E: Sub-process /usr/bin/dpkg returned an error code (1)

```
I think it's because I have a 2GB Jetson card, I tried to find out how to modify the image to match my card but being not very experienced I didn't succeed. 

From this link: https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/quick_start.html#wwpID0E0EB0HA
There is the variable $ (BOARD) that changes to jetson-nano-2gb-devkit . 

I don't know if this is the source of my problems, although I strongly suspect it. 

Could you help me on this ?

pythops, 2021-02-10:
For Jetson-nano 2GB, you need to edit the script `create-image.sh` in the line 54 and change it to from
```
./jetson-disk-image-creator.sh -o jetson.img -b jetson-nano -r 200
```
to 
```
./jetson-disk-image-creator.sh -o jetson.img -b jetson-nano
```
Nvidia Docs [here](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/flashing.html) section `Flashing to an SD Card`
</br></br>

For the DNS, try to restart `systemd-resolved`
```
$ sudo systemctl restart systemd-resolved.service
```

natdop, 2021-02-10:
It works, thank you! :) 
It was : 
`./jetson-disk-image-creator.sh -o jetson.img -b jetson-nano-2gb-devkit`

Unfortunately I still have the DNS problem after 
`$ sudo systemctl restart systemd-resolved.service`
I looked into this : https://github.com/pythops/jetson-nano-image/pull/15 but still I can't get it work
```
$ cat  /etc/systemd/resolved.conf

[Resolve]
DNS=1.1.1.1
FallbackDNS=8.8.8.8
#Domains=
LLMNR=no
MulticastDNS=no
DNSSEC=yes
DNSOverTLS=yes
Cache=yes
DNSStubListener=yes
ReadEtcHosts=yes
```

pythops, 2021-02-10:
I'm gonna update the scripts to support `Jetson Nano 2GB`.
Meanwhile, you can follow these steps to solve the DNS issue
```
$ sudo systemctl stop systemd-resolved.service
$ sudo systemctl disable systemd-resolved.service
$ echo "nameserver 1.1.1.1" | sudo tee  /etc/resolv.conf
$ sudo chatrr +i /etc/resolv.conf
```

natdop, 2021-02-10:
I got this : 
```

$ echo "nameserver 1.1.1.1" | sudo tee  /etc/resolv.conf
tee: /etc/resolv.conf : No such file or directory
```

But the file exist (in red with ls)
```

$ sudo chatrr +i /etc/resolv.conf
chattr: Opetation not supported while reading flags on /etc/resolv.conf
```

I really don't know what I did !

pythops, 2021-02-10:
You need to this 
```
$ sudo chattr -i /etc/resolv.conf
$ sudo rm -rf /etc/resolv.conf
$ echo "nameserver 1.1.1.1" | sudo tee /etc/resolv.conf
$ sudo chattr +i /etc/resolv.conf
```

natdop, 2021-02-10:
It will fix it even when I reboot ? 
I still got this : 
```
$ sudo chattr -i /etc/resolv.conf
chattr: Operation not supported while reading flags on /etc/resolv.conf
```
It's normal ?

pythops, 2021-02-10:
Try to remove it first 
```
$ sudo rm -rf /etc/resolv.conf
```

natdop, 2021-02-10:
Yes the other line after this worked

natdop, 2021-02-10:
Ok nice it's working even after a reboot, thank you for your help :)

pythops, 2021-02-10:
You're welcome :)

closed 2021-02-10 by pythops