pythops/jetson-image #21: Support jetson nano 2GB

pythops, 2021-02-12:


bunnyDrug, 2021-02-19:
What about the 2GB version requires special support?

I have the 2GB version so I am happy to test.

pythops, 2021-03-25:
Thanks @bunnyDrug

JunYa-88, 2021-05-03:
Is there any update on this? I am interested in this as well. In particular I would like to know what resources I should read to get started. I tried looking on the NVIDIA Jetson Linux Developer guide (https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/introduction.html), but it gives no clue on how to make custom images. @pythops any recommendations ?

bachmannn, 2021-05-13:
I tried on 2GB but it does not boot (getting bootloop with NVIDIA logo whitescreen then blackscreen).

Using the guide except changed username/password in ansible script.

pythops, 2021-05-15:
I made the necessary changes to support the jetson nano 2gb
Please take a look at the this PR #32 
Once 2 of you at least validated that this is working then I'll merge it to master and update the article in my blog.

sar, 2021-05-18:
> I made the necessary changes to support the jetson nano 2gb
> Please take a look at the this PR #32
> Once **2** of you at least validated that this is working (cause I don't have the 2gb board) then I'll merge it to master and update the article in my blog.

PR is working great on the jetson nano 2GB spec. Boots with only 110mb of allocated (active in process) memory consumption.

Did run into an issue with the filesystem throwing out of space errors:

```
$ ssh ...
-bash: cannot create temp file for here-document: No space left on device
$ sudo apt install -y cuda-toolkit-10-2 libcudnn8 libcudnn8-dev
Failed to fetch https://repo.download.nvidia.com/jetson/t210/dists/r32.5/InRelease  Error writing to output file - write (28: No space left on device)
```

```
$ fdisk -l
Disk /dev/mmcblk0: 29.86 GiB, 32044482560 bytes, 62586880 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt

Device          Start      End  Sectors  Size Type
/dev/mmcblk0p1  28672 62586846 62558175 29.8G Linux filesystem
/dev/mmcblk0p2   2048     2303      256  128K Linux filesystem
/dev/mmcblk0p3   4096     4991      896  448K Linux filesystem
/dev/mmcblk0p4   6144     7295     1152  576K Linux filesystem
/dev/mmcblk0p5   8192     8319      128   64K Linux filesystem
/dev/mmcblk0p6  10240    10623      384  192K Linux filesystem
/dev/mmcblk0p7  12288    13055      768  384K Linux filesystem
/dev/mmcblk0p8  14336    14463      128   64K Linux filesystem
/dev/mmcblk0p9  16384    17279      896  448K Linux filesystem
/dev/mmcblk0p10 18432    19327      896  448K Linux filesystem
/dev/mmcblk0p11 20480    22015     1536  768K Linux filesystem
/dev/mmcblk0p12 22528    22655      128   64K Linux filesystem
/dev/mmcblk0p13 24576    24959      384  192K Linux filesystem
/dev/mmcblk0p14 26624    26879      256  128K Linux filesystem

$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mtdblock0     31:0    0     4M  0 disk 
mmcblk0      179:0    0  29.9G  0 disk 
├─mmcblk0p1  179:1    0  29.8G  0 part /
├─mmcblk0p2  179:2    0   128K  0 part 
├─mmcblk0p3  179:3    0   448K  0 part 
├─mmcblk0p4  179:4    0   576K  0 part 
├─mmcblk0p5  179:5    0    64K  0 part 
├─mmcblk0p6  179:6    0   192K  0 part 
├─mmcblk0p7  179:7    0   384K  0 part 
├─mmcblk0p8  179:8    0    64K  0 part 
├─mmcblk0p9  179:9    0   448K  0 part 
├─mmcblk0p10 179:10   0   448K  0 part 
├─mmcblk0p11 179:11   0   768K  0 part 
├─mmcblk0p12 179:12   0    64K  0 part 
├─mmcblk0p13 179:13   0   192K  0 part 
└─mmcblk0p14 179:14   0   128K  0 part 
zram0        252:0    0 247.5M  0 disk [SWAP]
zram1        252:1    0 247.5M  0 disk [SWAP]
zram2        252:2    0 247.5M  0 disk [SWAP]
zram3        252:3    0 247.5M  0 disk [SWAP]
```

Modified `flash_image.sh` lines to flash on `/dev/sdb` where the SD card is located, extended the last partition.

```
# Extend fs
printf "\e[32mExtend the fs... "
e2fsck -fp $2"1" > /dev/null      # /dev/sdb1 instead of /dev/type0p1
resize2fs $2"1" > /dev/null       # /dev/sdb1 instead of /dev/type0p1
sync
printf "[OK]\e[0m\n"
```

closed 2021-05-19 by pythops