pythops/jetson-image #22: Ubuntu 18.04

Urtetoli, 2021-02-17:
what changes to make to install Ubuntu version 18.04?

Grungeby52, 2021-02-17:
> what changes to make to install Ubuntu version 18.04?

I was about to ask the exact same thing...

Best Regards

Onur Güzeldemirci

pythops, 2021-02-17:
You need to change `create-rootfs.sh` and the ansible tasks. 
I'm gonna update the scripts this week to support that.

Grungeby52, 2021-02-17:
Hi,

If I use this commit (Upgrade the ubuntu base image to 18.04.4) 

git checkout  a336c85d52dc580e42a77520bbed04228e63eca0

The image to be created is ubuntu 18.04 ?

Thanks for the quick reply

pythops, 2021-02-17:
@Grungeby52  that wouldn't work. more changes are needed.

pythops, 2021-02-20:
I made  decision to only support the latest stable version of Ubuntu.

closed 2021-02-20 by pythops