pythops/jetson-image #23: ./create-rootfs.sh: line 46: .../etc/resolv.conf: Operation not permitted

xxxcrow, 2021-02-19:
sudo -E ./create-rootfs.sh
Install the dependencies...  [OK]
Create rootfs directory...    [OK]
Run debootstrap first stage...  [OK]
./create-rootfs.sh: line 46: /home/ubuntu/nano-build/etc/resolv.conf: Operation not permitted

natdop, 2021-02-19:
It's maybe because of this commit : 
https://github.com/pythops/jetson-nano-image/commit/0964a02e1007a1e8f683d38ee33ec3f66a4ad53a
Try to git clone before this commit : 
https://github.com/pythops/jetson-nano-image/tree/89757c3c17d2b7fae039af3be995fa71e11b7d1e

pythops, 2021-02-19:
@xxxcrow 
FYI: when your use `chattr +i file` that makes the file immutable, that even root can not change.
What's OS you're using ?
Did you modify the `RELEASE` inside   `create-rootfs.sh` ?

xxxcrow, 2021-02-19:
@pythops whoah, i never even heard of any command like that, thank you, now i know 😃
I use raspberry pi 4 with ubuntu 20.04 😅

pythops, 2021-02-19:
I never tried to build the image on arm64.
I think some changes needed to be made if you wanna build it in raspberry pi 4 like changing the options for `debootstrap` in `create-rootfs.sh`  script.
Try to build it using a machine with x86_64 processor. 
I'm gonna close this issue as it's related to some unsupported build hardware.

closed 2021-02-19 by pythops

xxxcrow, 2021-02-20:
reinstalling ubuntu helped, but this time it was 20.10
i don't think it made a difference, but now everything works on the latest version of the script 🙄

pythops, 2021-02-20:
Well, good for you :))