pythops/jetson-image #24: Problem with dpkg and libegl1 package

technosar, 2021-03-20:
Build the image ...
patching file build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
Extract L4T...        dpkg: dependency problems prevent configuration of nvidia-l4t-core:
 nvidia-l4t-core depends on libegl1; however:
  Package libegl1 is not installed.

dpkg: error processing package nvidia-l4t-core (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 nvidia-l4t-core

The libegl1 package is already installed.

bunnyDrug, 2021-03-20:
Is this running on Ubuntu? If so, can you show the package installed via apt?

I feel, this issue is more to do with the nvidia apply debs script than it really has to do with this project.

add libegl1 and libexpat1 and try again:
Source: https://forums.developer.nvidia.com/t/customize-rootfs-dependency-error/122968/4

technosar, 2021-03-21:
Thank you for your answer.

The apt installation seems to be good 

libegl-mesa0/bionic-updates,now 20.0.8-0ubuntu1~18.04.1 amd64  [installed]
libegl1/bionic-updates,now 1.0.0-2ubuntu2.3 amd64  [installed]
libegl1-mesa/bionic-updates,now 20.0.8-0ubuntu1~18.04.1 amd64  [installed]
libegl1-mesa-dev/bionic-updates,now 20.0.8-0ubuntu1~18.04.1 amd64  [installed]
...
libexpat1/bionic-updates,bionic-security,now 2.2.5-3ubuntu0.2 amd64  [installed]
libexpat1-dev/bionic-updates,bionic-security,now 2.2.5-3ubuntu0.2 amd64  [installed, automatic]

ok, now i'm focusing on line 196 of nv-apply-debs.sh file :

LC_ALL=C chroot . dpkg -i --force-overwrite --path-include="/usr/share/doc/*" "${deb_list[@]}"

Thank you.

pythops, 2021-03-23:
@technosar 
Have you made any change to the `create-rootfs.sh` ? 
What were the exact steps that you followed before having this issue ?

technosar, 2021-03-24:
Where the archive :
http://us.archive.ubuntu.com/ubuntu/dists/focal/main/binary-arm64/Packages

technosar, 2021-03-24:
Please can you try sudo -E ./create-rootfs.sh

I have this error now : 
http://security.ubuntu.com/ubuntu/dists/bionic-security/main/binary-arm64/Packages  404  Not Found [IP : 91.189.91.39 80]

closed 2021-03-24 by pythops

bunnyDrug, 2021-03-24:
> Please can you try sudo -E ./create-rootfs.sh
> 
> I have this error now :
> http://security.ubuntu.com/ubuntu/dists/bionic-security/main/binary-arm64/Packages 404 Not Found [IP : 91.189.91.39 80]

I think you need to review your environment - that looks like an apt error getting your repositories. 
These issues you’re having are more fundamental and no defects with the project I’m afraid. 

does apt get update work for you?

technosar, 2021-03-25:
yes working now, i just remove arm64 architecture on the dpkg configuration.

husamia, 2021-04-26:
> yes working now, i just remove arm64 architecture on the dpkg configuration.

I removed the arm64 from the ./create-rootfs.sh then it won't complete creating the rootfs

```
sudo -E ./create-rootfs.sh
Install the dependencies...  [OK]
Create rootfs directory...    [OK]
Run debootstrap first stage...
```

rmoriz, 2021-12-18:
I'm getting the same error with the latest version of this git repo running on Ubuntu 20.04.2-LTS (amd64) inside vagrant.