pythops/jetson-image #25: How would you use this to distribute your own application for Jetson Nano

js1972, 2021-03-23:
I'm trying to work out how I would use this to distribute a DeepStream application I have developed for the Jetson Nano.

I need to install DeepStream over the Jetpack base and then install my application. Would this repo cater to that?

Would I just run `create-rootfs.sh` script then enter the chroot with `sudo chroot $JETSON_ROOTFS_DIR` and from that point I'm running in the new root directory structure and can just `apt install` all my dependencies and run my own installer script?

pythops, 2021-03-23:
follow  all the instructions to build the image and then flash your jetson nano board with this image.
Once you boot you board with the new image, log in and get  shell then install  your app as you would do with the jetpack image.

js1972, 2021-03-23:
Ah... so its still a manual install activity after first boot. I was thinking I could install my app and its dependencies on to an sd card and distribute the sd card.

pythops, 2021-03-23:
If you are app needs NVIDIA libraries that wouldn't work at this stage. 
If you need to distribute your app, I would suggest that you create a docker image for that, or you clone the SD card with your app installed in it and distribute it.

closed 2021-03-23 by pythops