pythops/jetson-image #26: JetPack 32.5

Gilmar-Gomes, 2021-03-31:
First of all, congratulations for this fabulous work!

Any chance to support JetPack 4.5 and L4T 32.5.0? I had several problems changing the BSP for this version, in create-image.sh. I can't figure out a solution. 

Thanks!

pythops, 2021-04-01:
Thanks @Gilmar-Gomes 
Yes sure, I'll try to  update to the latest BSP in the upcoming days.

pythops, 2021-04-02:
@Gilmar-Gomes 
The latest version of BSP is now supported

closed 2021-04-02 by pythops

pvmilk, 2021-05-29:
Thanks a lot for the work.
This really helps me shorten my startup time with the jetson nano. 

This is not really an issue report, but rather a question on the L4T and JetPack.

1.) Am I correct that L4T is more like the driver for the nano, and JetPack is like the ML/Container package of the nano?

2.) Am I correct that in your script, when you referred to BSP, it will install the L4T automatically, but not the JetPack?
(For the JetPack, I guess I can simply install with $sudo apt install nvidia-jetpack)

3.) Am I correct that the final image will not include GUI (i.e. ubuntu-desktop), and I can access with as a headless server (headless in term or GUI)?

Thanks in advances for the reply.

pythops, 2021-05-29:
> Am I correct that L4T is more like the driver for the nano, and JetPack is like the ML/Container package of the nano?

For more accurate definitions:

From https://developer.nvidia.com/embedded/linux-tegra, https://developer.nvidia.com/embedded/jetpack

L4T 

> NVIDIA L4T is the board support package for Jetson. It includes Linux Kernel 4.9, bootloader, NVIDIA drivers, flashing utilities, sample filesystem based on Ubuntu 18.04, and more for the Jetson platform.

Jetpack

> JetPack SDK includes the latest Jetson Linux Driver Package (L4T) with Linux operating system and CUDA-X accelerated libraries and APIs for Deep Learning, Computer Vision, Accelerated Computing and Multimedia. It also includes samples, documentation, and developer tools for both host computer and developer kit, and supports higher level SDKs such as DeepStream for streaming video analytics and Isaac for robotics.

------

> Am I correct that in your script, when you referred to BSP, it will install the L4T automatically, but not the JetPack?

Yes. My scripts are based on the latest l4T (linux kernel, bootloader...) that creates a minimalist image **without any additional tool**. I leave to each user the choice to pick up and install whatever they want on top of it.

-------

> Am I correct that the final image will not include GUI (i.e. ubuntu-desktop), and I can access with as a headless server (headless in term or GUI)?

Yes. there is an SSH server installed by default. the default user is `pythops` and the password is `pythops`. You can override those in this file `https://github.com/pythops/jetson-nano-image/blob/master/ansible/roles/jetson/defaults/main.yaml`