pythops/jetson-image #28: creating custom image directly on usb ssd attached to Jetson Nano

biocodz, 2021-04-07:
First of all, thanks for the great job creating these scripts and the instructions!

I'm trying to apply this to setup a new USB SSD attached to my Jetson Nano board, which currently runs NVIDIA supplied image (Ubuntu 18) from microSD.

It's surely should be possible to create the rootfs and the custom image Directly on the new USB SSD attached to the Jetson Nano i.e. without using another host. My doubts are about re-flashing the QSPI-NOR to the latest version, so that to enable booting from the USB SSD.

To summarize the goals:
Create custom image on the USB SSD attached to the Running Jetson Nano -> re-flash the QSPI-NOR -> reboot to USB SSD

pythops, 2021-04-08:
There are some modifications that you need to make to the scripts to make them run in jetson nano. for example, the line 52 in `create-rootfs.sh` should be removed. 
I don't think I can help as I don't have SSD attached to my jetson nano, and I always use my laptop to build the image for sdcard.
You can give it a try and if you succeed then you can create a PR about it

biocodz, 2021-04-08:
Thanks for the suggestion. I'll try experimenting.

closed 2021-04-08 by biocodz

biocodz, 2021-04-09:
I'm getting a slew of errors related to unmet dependencies because the Tegra packages were built for Ubuntu 18.04 that has many packages outdated in 20.10.
For example:
```
...
dpkg: dependency problems prevent configuration of nvidia-l4t-3d-core:
 nvidia-l4t-3d-core depends on libx11-6; however:
  Package libx11-6 is not installed.
 nvidia-l4t-3d-core depends on libxext6; however:
  Package libxext6 is not installed.
...
```
If I try fixing the dependencies using `chroot` and `apt --fix-broken install`, it installs versions of packages incompatible with the Tegra. I either have to re-build the Tegra packages or manually install all the required dependencies.

Any ideas as to why you didn't have such problems?

pythops, 2021-04-09:
Probably it's due to the version you're using of Ubuntu. I only tested with Ubuntu **20.04**