pythops/jetson-image #29: Jetson Nano 4GB

husamia, 2021-04-26:
I am trying to build the image for the Jetson Nano 4GB version. When I get to the create the image I get the error about the libegl1 not installed but it's indeed installed. 

```

~/jetson-nano-image$ sudo -E ./create-image.sh
Build the image ...
Download L4T...       [OK]
patching file /home/husamia/jetson-nano-image/build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
Extract L4T...        dpkg: dependency problems prevent configuration of nvidia-l4t-core:
 nvidia-l4t-core depends on libegl1; however:
  Package libegl1 is not installed.

dpkg: error processing package nvidia-l4t-core (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 nvidia-l4t-core
```

closed 2021-04-26 by husamia

oldgalileo, 2021-08-08:
@husamia did you ever solve this issue? Just encountered this.

husamia, 2021-08-09:
No. If you have a solution let us know.

oldgalileo, 2021-08-09:
Unfortunately, @pythops 's solution in issue #30 of running ansible-playbook did the trick. Running `sudo -E $(which ansible-playbook) jetson.yaml` apparently made some changes that fixed this issue so that running `create-image.sh` worked like a charm. 

We're running on 18.04 targeting a 20.04 image. Also worth adding we haven't actually successfully booted this yet because of flashing complications, so take this with a **_massive_** grain of salt. I'll come back and post with the results of the flashing in a day or two.