pythops/jetson-image #30: RE: https://github.com/pythops/jetson-nano-image/issues/24

husamia, 2021-04-26:
I am trying to build the image for the Jetson Nano 4GB version. When I get to the create the image I get the error about the libegl1 not installed but it's indeed installed. 

```

~/jetson-nano-image$ sudo -E ./create-image.sh
Build the image ...
Download L4T...       [OK]
patching file /home/husamia/jetson-nano-image/build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
Extract L4T...        dpkg: dependency problems prevent configuration of nvidia-l4t-core:
 nvidia-l4t-core depends on libegl1; however:
  Package libegl1 is not installed.

dpkg: error processing package nvidia-l4t-core (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 nvidia-l4t-core
```

so someone suggested to remove the amd64 from the file create-rootfs.sh which I did but now it won't complete the process like before it stops
```

sudo -E ./create-rootfs.sh
Install the dependencies...  [OK]
Create rootfs directory...    [OK]
Run debootstrap first stage...

```

pythops, 2021-04-26:
Don't build the image inside the jetson board. The current scripts requires a machine with x86-64 arch to work.

closed 2021-04-26 by pythops

husamia, 2021-04-27:
> Don't build the image inside the jetson board. The current scripts requires a machine with x86-64 arch to work.

I am not building the image inside the jetson board. It's being built on a WSL2 Ubuntu. Do you have any suggestions?

pythops, 2021-04-29:
Did you run  `sudo -E $(which ansible-playbook) jetson.yaml` ?  what was the output ?

husamia, 2021-04-29:
> Did you run `sudo -E $(which ansible-playbook) jetson.yaml` ? what was the output ?

this is what I get 
```

~/jetson-nano-image/ansible$ sudo -E $(which ansible-playbook) jetson.yaml

PLAY [Customize Jetson rootfs] 
TASK [Gathering Facts] fatal: [jetson]: FAILED! => {"msg": "jetson is not a directory"}

PLAY RECAP *jetson                     
: ok=0    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0
```

pythops, 2021-04-30:
Just follow the instructions here
https://pythops.com/post/create-your-own-image-for-jetson-nano-board.html