pythops/jetson-image #31: Cannot run Ansible Playbook jetson.yaml - Failed to update apt cache unknown reason

bachmannn, 2021-05-13:
Building on Ubuntu 20.04 with latest updates but `ansible-playbook` fails. Does ansible run this inside some sort of VM on the `rootfs` and apt can't access the internet or some other issue causing below?

```
$ export JETSON_ROOTFS_DIR="/tmp/jetson/rootfs"
$ cd ansible/
$ sudo -E ansible-playbook -vvv jetson.yaml

# fails on this step, no verbose output

TASK [jetson : Upgrade packages ] *******************************
fatal: [jetson]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: unknown reason"}
```

pythops, 2021-05-13:
Some issue with `apt` not being able to update the cache. Probably a network issue.
Try again with the same command, and if it does fail then try to chroot into the jetson roortfs and run `apt update` manually
```
$ sudo chroot /tmp/jetson/rootfs
# apt update 
```

bachmannn, 2021-05-13:
> ```
> $ sudo chroot /tmp/jetson/rootfs
> # apt update 
> ```

I had to change `/rootfs/etc/resolv.conf` to `8.8.8.8` because nameservers were set to defaults `nameserver 1.1.1.1` after running `create-rootfs.sh` which is invalid, not sure where debootstrap got these values from?

Then it worked, :+1:  Issue can be closed now.

pythops, 2021-05-13:
FYI: `nameserver 1.1.1.1` is set in `create-rootfs.sh` 
https://github.com/pythops/jetson-nano-image/blob/master/create-rootfs.sh#L46

closed 2021-05-13 by pythops