pythops/jetson-image #33: Tutorial Missing Step?

lavachemist, 2021-05-18:
Hello,

I followed the tutorial a few times and always got stuck at ```sudo -E $(which ansible-playbook) jetson.yaml```

Turns out that I also needed to install ansible-base. I know this isn't an issue with the repo, but I didn't see another place that I could bring it up. It might help other users too.

sar, 2021-05-19:
If you're on Ubuntu and `pip3 install --user ansible` has missed dependencies (for whatever reason), you can grab it from default repos `sudo apt install -y ansible`.

lavachemist, 2021-05-19:
> If you're on Ubuntu and `pip3 install --user ansible` has missed dependencies (for whatever reason), you can grab it from default repos `sudo apt install -y ansible`.

good suggestion.

The other issue I had was that the flash-image.sh script failed due to lines 68 and 69:

```e2fsck -fp $2"p1" > /dev/null
resize2fs $2"p1" > /dev/null```

My device was at /dev/sdb, but it tried to resize sdbp1. I changed it to this, and it worked:

```e2fsck -fp $2"1" > /dev/null
resize2fs $2"1" > /dev/null```

pythops, 2021-05-19:
@sar 
Thanks for your help, I appreciate

@lavachemist 
Usually, the sdcard are recognized under the name `mmcblk0` and for each partition there is `pX`, example (`p1` for the 1se partition, `p2` for the second and so on). If your system, for some reason, does not follow this pattern, then indeed you need to change it as you showed.

I personally do not recommend install pip with sudo or use apt. I would recommend using pip as normal user. 
Can you paste here the errors you had when trying to install ansible ?

closed 2021-05-22 by pythops