pythops/jetson-image #36: Cannot install nvidia-jetson with Ubuntu Focal with BSP r32.5.1

pvmilk, 2021-06-03:
```
$ sudo apt-get install nvidia-jetpack

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 nvidia-jetpack : Depends: nvidia-opencv (= 4.5.1-b17) but it is not going to be installed
                  Depends: nvidia-tensorrt (= 4.5.1-b17) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
Could this be because BSP r32.5.1 haven't supported Ubuntu Focal yet?

I tried to install `nvidia-tensorrt` and the outcome was the unmet version of python3. 

P.S Actually, I don't really need nvidia-jetpack in my setup. I was trying to solve my [issue on OpenGL](https://github.com/pythops/jetson-nano-image/issues/35), so I was trying to install many related packages to test, but found this by accident.

pvmilk, 2021-06-04:
Found the answer here. 

https://forums.developer.nvidia.com/t/when-will-jetpack-move-to-ubuntu-20-04/142517

pythops, 2021-06-05:
If you need to install `tensorrt` you can do it with the following commad
```
$ sudo apt install tensorrt
```
I'm gonna close this issue and if you need to discuss how to install some software on the top of the minimalist image that you build with this repo, I would suggest that you use the `Disucssion` section in github for that

closed 2021-06-05 by pythops