pythops/jetson-image #44: Create img for Ubuntu 18 bionic

vcasadei, 2021-07-10:
Hi, I find your project great, however, I need to use Ubuntu 18 and I've not been able to do it. I tried to change the ansible to choose bionic and RELEASE=bionic on create-rootfs.sh.

However, the image continues to be 20.04. I think it's something about the create-image.sh, but I do not know... Could you help?

pythops, 2021-07-10:
I don't support ubuntu 18.04 anymore. however you can go back through commits history and you'll find the changes I made to move from 18.04 to 20.04

closed 2021-07-10 by pythops

eusoubrasileiro, 2022-05-12:
well for those who think that Bionic might be useful the starting commit is [b4a6abd11fba717f48a67a43c2a501589c709163 ](https://github.com/pythops/jetson-nano-image/commit/b4a6abd11fba717f48a67a43c2a501589c709163)

Just want to warn that if you are getting problems (like me) training to installing something else be careful. The problem might be the missing installation of  JetPack or Deepstream. 

@pythops please consider including these links on the end of your tutorial. [JetPack](https://developer.nvidia.com/embedded/jetpack) or [Deepstream](https://developer.nvidia.com/deepstream-getting-started). They are required for a lot of packages or tools on jetson-nano. 

@pythops anyway thanks a lot for your code.

pythops, 2022-05-13:
@eusoubrasileiro 
Thanks for the feedback.
I try not to include any package on top of the image as I want to keep it very minimalist. 
I'll open an issue to let people add the packages they mostly struggle to install and I'll add tutorials on how to install them

pythops, 2022-05-13:
https://github.com/pythops/jetson-nano-image/issues/62

eusoubrasileiro, 2022-05-19:
Managed to succeed but did not find time to customize and adapt it to make a new PR. For reference [here](https://github.com/eusoubrasileiro/jetson-nano-image/tree/bionic_latest).  Everything now working finally. 

I followed the nvidia scripts strictly. Code based totally on scripts from BSP `tools\samplefs` folder. 
-  `nv_build_samplefs.sh`
- `nvubuntu_samplefs.sh`.