pythops/jetson-image #49: Jetson Nano 4GB B01 Stuck at Nvidia Logo Screen

KamalanathanN, 2021-09-28:
hi, i followed your instructions from https://pythops.com/post/create-your-own-image-for-jetson-nano-board.html
everything was working as explained in it exept it is stuck on boot showing only the nvidia logo.
after flashing with your image i'm now unable to even make the official nvidia images working 😭 it too gets stuck on boot.
please help. 🥺

vkuehn, 2021-09-28:
Hi, I had the same issue. To resolve it I have used another sd card, removed Wifi, added a more robust AC adapter. Not sure what the issue was

bunnyDrug, 2021-09-28:
If @vkuehn ‘s suggestion doesn’t work you may need to flash the board. 

http://docs.nvidia.com/sdk-manager/%20install-with-sdkm-jetson/index.html

KamalanathanN, 2021-09-28:
> If @vkuehn ‘s suggestion doesn’t work you may need to flash the board.
> 
> http://docs.nvidia.com/sdk-manager/%20install-with-sdkm-jetson/index.html

I'm currenty using Ubuntu 20.04 so NvidiaSDK manager shows No available releases for 20.04.
are you suggesting i downgrade to 18.04, reflash to the one that worked for me which is Jetpack 4.5.1(hope it works)and then try the instructions again?

pythops, 2021-09-28:
follow @vkuehn suggestion it should work

closed 2021-09-28 by pythops

bunnyDrug, 2021-09-28:
> I'm currenty using Ubuntu 20.04 so NvidiaSDK manager shows No available releases for 20.04. are you suggesting i downgrade to 18.04, reflash to the one that worked for me which is Jetpack 4.5.1(hope [...]

So, the requirements for the skd manager states that 20.04 is supported.
However, I’d 100% try @vkuehn solution before embarking on this.