pythops/jetson-image #52: Support for Xavier NX soon

aditbhrgv, 2022-01-10:
Hi,

Thank you for your work.
Can you please let me know how can I make this repo work for Jetson Xavier NX ?

Thanks

aditbhrgv, 2022-01-17:
Thanks for marking it as enhancement. 
Could you please let me know how can I use the flash script to flash to eMMC?

Thanks

MauiJerry, 2022-05-28:
also the Xavier AGX please.
I did find instructions on the SAT-Immersive github site for (re)building for NX https://gitlab.com/sat-metalab/distributions/jetson-images

pythops, 2022-05-28:
@MauiJerry 
I don't have the Xavier AGX board, so it's gonna be little bit hard for me add support for it. Unless someones send it to me, then I'll do it

MauiJerry, 2022-05-28:
im working thru it. the SAT folks got it running on nx but with older sdk
etc. im looking into magic words to put in their patch scripts. the 5.0 sdk
looks nice.
ill share results.
pity nvidia doesn't send you hardware to get free-ish labor

- jerry isdale

sent from my phone, which may have added its own interpretation of what i
typed or dictated. I apologize for any auto correct errors introduced by
our robotic under lord.

pythops, 2022-10-15:
Unless someone send me the boards or contribue with a PR then I have to close that ticket unfortunately

closed 2022-10-15 by pythops