pythops/jetson-image #64: CreateRootFS is taking a very long time

rem0nz, 2022-05-15:
debootstrap in createrootfs.sh is extremely slow. It takes more than 12 hours to finish. 
Anything I can do to fix that ?

pythops, 2022-05-15:
That's not normal. It should take couple of a minutes not hours ! 
Something is wrong with your machine, it could your CPU or the network is too slow ...

closed 2022-05-15 by pythops

rem0nz, 2022-05-15:
Actually no , both of them are good Intel i7 & Ethernet with 600mbps . 
I found out that when I ran the same code on a Jetson Nano (arm64) it was working fine, but on the Intel processor , it was very slow. 
Could be emulation problem ? this is a brand new ubuntu 18.04 installation.

pythops, 2022-05-15:
:thinking: that's strange.
Try to to use the build  environment that I mentioned in the blog post (Virtualbox + vagrant)
I don't think  I can go further than that as it's related to your setup, sorry.