pythops/jetson-image #71: The gpu drivers seems doesn't working correctly .

mrcmunir, 2022-08-22:
If installed any GUI (Mate , ubuntu , unity , lubuntu , lxqt)  with GDM manager is stuck in the login with normal user
Only working with root user .
With lightdm can login but segment fault in any sample .
Is there a package or symbolic link missing to work correctly?

mrcmunir, 2022-08-24:
For the configure usb service mode service failed it's due missing /sbin/brctl needed bridge-utils install .
As for the acceleration, as soon as I have time, I'll look at it if I can fix it.

pythops, 2022-08-29:
The GUI issue seems to be out of scope as this repo is about the base image. 

> For the configure usb service mode service failed 
Can you explain more here

mrcmunir, 2022-08-30:
> The GUI issue seems to be out of scope as this repo is about the base image.
> 
Regarding the interface, I have solved it by going down to 18.04 base for now.
> > For the configure usb service mode service failed
> 
> Can you explain more here
There is a service that starts every time configure usb service mode that autostart fails nv-l4t-usb-device-mode.service
Install bridge-utils and it doesn't complain anymore

pythops, 2022-10-15:
Gonna look to it, thanks

pythops, 2022-10-15:
I added the `birdge-utils` package. Thanks for your reporting :pray:

closed 2022-10-15 by pythops