pythops/jetson-image #73: Rootfs Error on Install the Dependencies

MrZahl, 2022-08-31:
[image]
Hello, can you help me with this problem? Is said that can't install because is not signed.

Thank you,

HrZahl.

pythops, 2022-08-31:
You seems to have Debian operating system right ? You need to use a VM with ubuntu to processed. Check the `# Build environment (Optional)` section in the tutorial

MrZahl, 2022-09-01:
[image]
Hello, no, i'm using ubuntu...

pythops, 2022-09-01:
check your `/etc/apt/sources.list`. Somehow you can not access a link (401 error)  in your config. You can comment it and run `$ sudo apt update`. 
Gonna close that one as it's not related to this repo but to your host config instead.

closed 2022-09-01 by pythops