pythops/jetson-image #74: updating to BSP version 35.1 results in "no space left on device"

aditbhrgv, 2022-11-01:
Hi,

have  you already tested the latest BSP jetson_linux_r35.1.0_aarch64.tbz2? We use Ubuntu 20.04 as a base and can build an image for Jetson with https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t186/jetson_linux_r32.6.1_aarch64.tbz2. On the other hand using the https://developer.nvidia.com/embedded/l4t/r35_release_v1.0/release/jetson_linux_r35.1.0_aarch64.tbz2 always throws an error due to missing space. 

Are you aware of this issue ?

[   1.5014 ] Warning: pub_key.key is not found
[   1.5026 ] tegrahost_v2 --chip 0x19 0 --updatesigheader 1_Image_aligned_sigheader.encrypt 1_Image_aligned_sigheader.hash zerosbk
Traceback (most recent call last):
  File "/home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/tegraflash.py", line 1369, in <module>
    tegraflash_run_commands()
  File "/home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/tegraflash.py", line 1210, in tegraflash_run_commands
    interpreter.onecmd(command)
  File "/usr/lib/python3.8/cmd.py", line 217, in onecmd
    return func(arg)
  File "/home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/tegraflash.py", line 734, in do_sign
    self.chip_inst.tegraflash_encrypt_sign_binary(exports, args)
  File "/home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 1016, in tegraflash_encrypt_sign_binary
    shutil.copyfile(new_filename, paths['WD'] + "/" + new_filename)
  File "/usr/lib/python3.8/shutil.py", line 275, in copyfile
    _fastcopy_sendfile(fsrc, fdst)
  File "/usr/lib/python3.8/shutil.py", line 166, in _fastcopy_sendfile
    raise err from None
  File "/usr/lib/python3.8/shutil.py", line 152, in _fastcopy_sendfile
    sent = os.sendfile(outfd, infd, offset, blocksize)
**OSError: [Errno 28] No space left on device: 'Image_sigheader.encrypt' -> '/home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/mnt/Image_sigheader.encrypt'**
l4t_sign_image.sh: Error: Unable to find the signed file generated by tegraflash.py
failed.

**Base is a vm with vagrant as proposed. The device with no space left seem to be the system.img**

**/dev/loop5       3287168   3139244         0 100% /home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/mnt**
xxx@jetson:~/icmore-jetson$ sudo losetup /dev/loop5
/dev/loop5: [2049]:779949 (/home/xxx/JETSON_BUILD_DIR/Linux_for_Tegra/bootloader/system.img)

pythops, 2022-11-01:
BSP 35 does not support jetson nano, only agx and orin boards.
What you have in this repo is the latest you can have.

closed 2022-11-01 by pythops

aditbhrgv, 2022-11-01:
We have already been using Jetson Xavier NX board with BSP 32.6.1. 
So, we thought updating BSP to 35.1 for Xavier NX should be supported straight-forwarded.

Do you have any plans to update this repo for Xavier NX ?

pythops, 2022-11-01:
I need a board to test it before saying it's officially supported but if you want to run the tests then I'll start working on it.