pythops/jetson-image #79: Update To LT4 32.7.3?

ByerRA, 2023-01-30:
Is there any plans to update your scripts to use LT4 32.7.3?

pythops, 2023-01-30:
Sure, this week ;)

pythops, 2023-03-20:
Done

ByerRA, 2023-03-26:
Thanks for the update.

You need to add "makedb" and "jq" to the list of required packages on your documentation page as from a fresh Ubuntu install those packages are not installed and are required for your package to execute.

pythops, 2023-03-26:
`jq` yes cause I use in `scripts/build-base-rootfs.sh` but why `makedb` ?

ByerRA, 2023-03-28:
> `jq` yes cause I use in `scripts/build-base-rootfs.sh` but why `makedb` ?

Sorry for the late reply, been busy.  When I executed the "just build-jetson-rootfs" command I got an error about "makedb" missing so I added it and everything worked from there.