pythops/jetson-image #81: Jetson Nano can't boot when all 3 .sh files run correctly!

morteza-h92, 2023-02-14:
dear community
when I done step by step your instruction in link:
https://pythops.com/post/create-your-own-image-for-jetson-nano-board.html 
correctly, My Jetson Nano can't boot after insert SD card (screen of monitor can not show any Nvidia logo, just on with black screen and after 30sec restart again).
result of .sh files are same as you but to me, some files could be corrupted when image file flashed in jetson. 
My framework:
host : ubuntu 20.04 amd64
target : jetson nano 2g

any help would be appreciated
best regards

pythops, 2023-02-14:
Try again to flash the image or maybe try another sd card

morteza-h92, 2023-02-15:
> Try again to flash the image or maybe try another sd card

1)result of flash-image.sh file : 
sudo ./flash-image.sh jetson.img /dev/sdd
Unmount SD card... [OK]
2831155200 bytes (2.8 GB, 2.6 GiB) copied, 167 s, 16.9 MB/s
676+1 records in
676+1 records out
2837446656 bytes (2.8 GB, 2.6 GiB) copied, 175.987 s, 16.1 MB/s
[OK]
Extend the partition... [OK]
Extend the fs... resize2fs 1.45.5 (07-Jan-2020)
[OK]
Success!
Your sdcard is ready!
after boot jetson with new flashed sd card, screen of monitor can not show any Nvidia logo..
seems that flash procedure correctly done but why jetson can't boot? 
can you show me result of your flashing image when boot jetson?

2) also format sd card (ext4) and try again with balena etcher:
https://www.balena.io/etcher
to flash with another tool, but results are same and screen can not show nvidia logo..

3) create-rootfs.sh and create-image.sh files run correctly. 
note that my host computer is amd64 ubuntu 20.04 but in creat-rootfs.sh file :  
ARCH=arm64
this variable isn't same as my host arch and when i change to amd64, get error in download from package manager in create-image.sh because jetson can not know arch amd64.  this variable, ARCH, should be arm64 or change?

4) you can see my flashed image in attached files, some directories have corrupted files. this is normal? 

[image]
[image]
[image]
[image]
[image]

pythops, 2023-02-15:
You don't need any flash tool, `./flash-image.sh` does it.
the image I show is by runnng `htop`
This issue seems to be related to something else (your board or sdcard)

closed 2023-02-15 by pythops