pythops/jetson-image #83: Facing issues with UART

UtkarshVerma, 2023-02-18:
Hello, firstly thanks for your excellent work on this project.

I have the Jetson Nano 2GB variant and everything works fine on the board. However, I seem to be facing issues with accessing the UART port. I can't get the loopback test to work on `/dev/ttyTHS1` (Pins 8,10 on the 40-pin connector). I have disabled the `nvgetty` service as well.

Could you please let me how I can use the UART pins?

Thanks and regards,
Utkarsh

pythops, 2023-02-18:
I never used the UART port on jetson nano. I usually connect to it through ssh. 
I can give it a try when I have some time. Meanwhile I'll let this issue open if someone from the community has already the answer

pythops, 2023-03-20:
I'll close this issue as there is no update for it.

UtkarshVerma, 2023-03-26:
One thing I noticed is that this issue is only specific to Jetson Nano 2GB. On the 4GB variant, I didn't face this issue.