pythops/jetson-image #84: Use podman instead of Virtualbox

pythops, 2023-03-07:


bdh1993, 2023-03-17:
## How to build it?

I cloned the source.
cd jetson-nano-image
./scripts/build-base-rootfs.sh
./scripts/build-jetson-image.sh

## error from build-jetson-image.sh file

The following error message was received during the build.

**Error: invalid config provided: CapAdd and privileged are mutually exclusive options**

---
## Build succeeded, but jetson.img is not available

Delete the "--cap-add=all" option and build.tar file and rebuild.
Successfully created "jetson.img".
But not opened jetson.img and not used jetson nano (a2)

---
## Is there a problem if the "--cap-add=all" option is not used?

pythops, 2023-03-17:
there is a WIP PR #85

closed 2023-03-20 by pythops