pythops/jetson-image #91: User permissions and nvmpi decoder failing (jetson-ffmpeg)

eusoubrasileiro, 2023-04-13:
After using the scripts back again after one year or so I ended up with bootable image. 
Some little problems:

1 .The home directory is owned by root not by the user created. 
      It seams, don't know why Containerfile.rootfs  `useradd` isn't working well maybe because everything is copied inside `/build/Linux_for_Tegra`? I don't know... 
2. The same as 1. for /tmp folder. 
3. Canot make jetson-ffmpeg work again..[ exactly as this previous issue on an old version from hre](https://github.com/pythops/jetson-nano-image/issues/62#issuecomment-1126112501)
```Error: Can't initialize nvrm channel Error: 
Can't initialize nvrm channel 
Couldn't create ddkvic Session: 
Cannot allocate memory nvbuf_utils: 
Could not create Default NvBufferSession  
```
Maybe this weird behaviour is somehow related to nvidia not officially supporting focal only bionic? Who knows?

For while this is it. Will keep my tests and reporting here... I am certainly going to bionic again, just wondering if podman could create something usefull (without the errors above) without chroot, like you did.

All in all thanks a lot for introducing me to podman and just. 
Best Regards @pythops!

eusoubrasileiro, 2023-04-14:
solved the 3rd one from [here](https://github.com/jocover/jetson-ffmpeg/issues/119#issue-1235765653) and more from the script `l4t_create_default_user.sh` when creating user groups should be `-G "sudo,video,audio,adm"`

pythops, 2023-04-14:
Thanks for reporting those points, I'll check them 
Let me know if you noticed other issues or struggling with some tools,

eusoubrasileiro, 2023-04-18:
@pythops thanks for the answer. 
I have tried a lot of things since your previous releases with vagrant...
I am still exploring but if possible I would like to kindly make two questions (interconnected), if you dont mind answering:

1. Why did you drop a isolated ubuntu (with vagrant) and chroot to use podman (without chroot)?
2. Why not use the scripts from nvidia BSP and chroot to create the image directly inside podman or docker? 

Thanks a lot!

eusoubrasileiro, 2023-04-18:
Forgot to mention ` just build-jetson-image ...`  asks for my sudo password is it expected or something I am doing wrong?

eusoubrasileiro, 2023-04-18:
the resize script (service) is not working on first boot here is the journalctl of the first boot 

> Mar 02 13:00:42 localhost resizerootfs.sh[5070]: GPT PMBR size mismatch (3706879 != 122909695) will be corrected by w(rite).
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Checking that no-one is using this disk right now ... FAILED
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: This disk is currently in use - repartitioning is probably a bad idea.
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Umount all file systems, and swapoff all swap partitions on this disk.
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Use the --no-reread flag to suppress this check.
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Disk /dev/mmcblk0: 58.6 GiB, 62929764352 bytes, 122909696 sectors
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Units: sectors of 1 * 512 = 512 bytes
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Sector size (logical/physical): 512 bytes / 512 bytes
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: I/O size (minimum/optimal): 512 bytes / 512 bytes
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Disklabel type: gpt
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Disk identifier: 9C1D46E6-7C0F-4281-9888-D1F3537D57B0
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Old situation:
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Device          Start     End Sectors  Size Type
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p1  28672 3702783 3674112  1.8G Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p2   2048    2303     256  128K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p3   4096    4991     896  448K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p4   6144    7295    1152  576K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p5   8192    8319     128   64K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p6  10240   10623     384  192K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p7  12288   13055     768  384K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p8  14336   14463     128   64K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p9  16384   17279     896  448K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p10 18432   19327     896  448K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p11 20480   22015    1536  768K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p12 22528   22655     128   64K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p13 24576   24959     384  192K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p14 26624   26879     256  128K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Partition table entries are not in disk order.
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p1:
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: New situation:
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Disklabel type: gpt
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Disk identifier: 9C1D46E6-7C0F-4281-9888-D1F3537D57B0
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Device          Start     End Sectors  Size Type
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p1  28672 3706846 3678175  1.8G Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p2   2048    2303     256  128K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p3   4096    4991     896  448K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p4   6144    7295    1152  576K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p5   8192    8319     128   64K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p6  10240   10623     384  192K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p7  12288   13055     768  384K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p8  14336   14463     128   64K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p9  16384   17279     896  448K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p10 18432   19327     896  448K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p11 20480   22015    1536  768K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p12 22528   22655     128   64K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p13 24576   24959     384  192K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: /dev/mmcblk0p14 26624   26879     256  128K Linux filesystem
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: Partition table entries are not in disk order.
Mar 02 13:00:42 localhost resizerootfs.sh[5070]: GPT PMBR size mismatch (3706879 != 122909695) will be corrected by w(rite).
Mar 02 13:00:42 localhost systemd[1]: resizerootfs.service: Main process exited, code=exited, status=1/FAILURE
Mar 02 13:00:42 localhost systemd[1]: resizerootfs.service: Failed with result 'exit-code'.

eusoubrasileiro, 2023-04-18:
Just downloaded and flashed your latest release (user jetson) the errors are the same. 

So I'm here sharing my [`journalctl > text.file`](https://gist.github.com/eusoubrasileiro/d6adad77e85b977eb0b244e29bb4ef49) for you in case it's useful.

pythops, 2023-04-18:
Good points. Let me answer them in the order

> Why did you drop a isolated ubuntu (with vagrant) and chroot to use podman (without chroot)?

The vagrant + virtualbox is too heavy . `podman` in the other hand is lightweight and provide the same solution for this project

> Why not use the scripts from nvidia BSP and chroot to create the image directly inside podman or docker?

That's what I'm doing now :) 

> `just build-jetson-image ...`  asks for my sudo password is it expected or something I am doing wrong?

You're doing anything wrong, it's normal that it asks for password cause rootful podman is needed. If you don't want to be asked for password for `podman` , you can add this line in your `/etc/sudoers`
```
<YOUR USERNAME HERE> ALL=(ALL) NOPASSWD:SETENV: /usr/bin/podman
```

> `resizerootfs` and `dns` errors

I'll check those and I'll fix them

eusoubrasileiro, 2023-07-24:
Thanks @pythops

pythops, 2023-11-24:
fixed here #103

closed 2023-11-24 by pythops