pythops/jetson-image #92: DNS issue

pythops, 2023-04-18:


ByerRA, 2023-04-21:
Having an issue with DNS using your process.

Now, I only use your scripts to build the "rootfs" because all my Jetson Nano's are 16G eMMc models and I can't flash the image created with your scripts.  So after I use your scripts to build the "rootfs" I then do the following to flash to the eMMc...

```shell
$ wget https://developer.nvidia.com/downloads/remetpack-463r32releasev73t210jetson-210linur3273aarch64tbz2 
$ tar -vxpf remetpack-463r32releasev73t210jetson-210linur3273aarch64tbz2 
$ rm remetpack-463r32releasev73t210jetson-210linur3273aarch64tbz2 
$ rm -rf ./Linux_for_Tegra/rootfs 
$ sudo cp -Rvp ./rootfs ./Linux_for_Tegra 
$ sudo chmod 4755 ./Linux_for_Tegra/rootfs/usr/bin/sudo 
$ patch ./Linux_for_Tegra/nv_tegra/nv-apply-debs.sh < ./patches/nv-apply-debs.diff 
$ cd ./Linux_for_Tegra 
$ sudo ./apply_binaries.sh
$ sudo ./flash.sh jetson-nano-emmc mmcblk0p1
```

The issue is that it's not picking up the DNS server from my networks DHCP.  If I look at the "/etc/resolv.conf" file the name server is set to "127.0.0.0" and while I can manually edit the "/etc/resolv.conf" file to use another name server, it won't keep after reboot and DNS isn't working.

I check the "/etc/resolv.conf" file in the "rootfs" directory your script created and it's there and proper but no DNS.  If I manually edit the "/etc/resolv.conf" file on the Jetson Nano, it's overwritten on reboot

The only way I could get DNS to work properly was to do the following on the Jetson Nano with DNS issues.

```shell
$ sudo rm -rf /etc/resolv.conf
$ sudo nano /etc/resolv.conf (adding "nameserver 8.8.8.8" to the file)
$ sudo apt-get update && sudo apt-get install resolvconf
$ sudo reboot 
```

And  this "does" fix the DNS issue and upon reboot it's picking up the proper DNS server from my DHCP server.

pythops, 2023-04-22:
One way to get it work without installing resolvconf is to override the `/etc/resolv.conf` and make it immutable so it does not get overriden 
```
$ echo "nameserver 8.8.8.8" | sudo tee /etc/resolv.conf
$ sudo chattr +i /etc/resolv.conf
```

I'm working on the new jetson nano orin and I'll push some fixes to fix dns among other stuff

ByerRA, 2023-04-24:
> One way to get it work without installing resolvconf is to override the `/etc/resolv.conf` and make it immutable so it does not get overriden
>
> ```
> $ echo "nameserver 8.8.8.8" | sudo tee [...]

Thanks, that solved the DNS issue, can't wait for your fixes :)

jruhland-mtm, 2023-05-05:
I confirmed via resolvectl that systemd is correctly identifying my network's DNS server on the eth0 interface, however this was never making it into the resolv.conf file.

Solved by replacing resolv.conf with a link to the one generated by systemd:
```
sudo rm /etc/resolv.conf
sudo ln -s /run/systemd/resolve/stub-resolv.conf /etc/resolv.conf
```

TheRoam, 2023-05-14:
hi. this affects **apt-get** and is a must do right after first boot! should be indicated somewhere in the readme. thanks!

pythops, 2023-11-24:
fixed here #103

closed 2023-11-24 by pythops