pythops/jetson-image #98: No clue of what revision number I have, No DNS, no Space Left alfter install

asophila, 2023-05-28:
Hi, 
I've followed an older way to install this minimal Jetson Image without problem (sort of long and cumbersome, but ok).
This new way using portainer and just is pure genius!

but:
3 problems:
1) How do I know my board revision number?  I know I have a B01 jetson Nano 4Gb. Is there a way to know the revision number?
2) I'm installing on a 128Gb card, and the system cries about having little space left. The manual doesn't explain how to expand the filesystem, is it ok to do it by force using "parted"?
3) DNS not working, cant install anything. I've tried a lot of stuff. Nothin works, I don't know why is this, Is this necessary for the stability of the system? if that's de case, then I'm unable to install cuda and cuda stuff.

Thanks for this. We (jetsonians) really appreciate what you do.

D3Nd3R, 2023-06-08:
Hi 
1. for B01 jetson Nano 4Gb I use 300 version 
2. to resolve space problem I start script: sudo /usr/local/bin/resizerootfs.sh . It should start after reboot but don't
3. to resolve DNS problem use  echo "nameserver 8.8.8.8" > /etc/resolv.conf  as superuser

pythops, 2023-11-24:
Fixed here #103

closed 2023-11-24 by pythops