pythops/jetson-image #100: Manjaro Exec format error

nullx1337, 2023-09-13:
Hello, 

i want to build the image on a Manjaro Vm running: 
```sh
DISTRIB_ID="ManjaroLinux"
DISTRIB_RELEASE="23.0.1"
DISTRIB_CODENAME="Uranos"
DISTRIB_DESCRIPTION="Manjaro Linux"
```

but i'll get following error
```sh
just build-jetson-rootfs  ✔
Building base rootfs
STEP 1/12: FROM ubuntu:focal
STEP 2/12: LABEL org.opencontainers.image.authors="Badr @pythops"
STEP 3/12: RUN apt update && apt install -y systemd
exec container process /bin/sh: Exec format error
Error: building at STEP "RUN apt update && apt install -y systemd": while running runtime: exit status 1
```
i've installed qemu-user-static.
Do i miss something, to build the arm64 arch?

nullx1337, 2023-09-13:
Oh sorry totaly forgot that, i updated the post.

pythops, 2023-09-13:
maybe you need to run those instructions (inside the VM of course) before running `just build-jetson-rootfs`
```
sudo apt install -y podman qemu-user-static lrzip
sudo systemctl start systemd-binfmt.service
 ```
 You can check the instructions order by looking at the CI flow
 https://github.com/pythops/jetson-nano-image/blob/master/.github/workflows/ci.yaml

nullx1337, 2023-09-13:
I run 
```sh
sudo pacman -S podman qemu-user-static lrzip 
Warning: podman-4.6.2-1 is up to date – Reinstall
Warning: qemu-user-static-8.1.0-2 is up to date – Reinstall
Warning: lrzip-0.651-2 is up to date – Reinstall
```
then
```sh
sudo systemctl start systemd-binfmt.service

sudo systemctl status systemd-binfmt.service 
systemd-binfmt.service - Set Up Additional Binary Formats
Loaded: loaded (/usr/lib/systemd/system/systemd-binfmt.service; static)
Active: inactive (dead)
Condition: start condition failed at Wed 2023-09-13 17:01:51 CEST; 3s ago
├─ ConditionDirectoryNotEmpty=|/lib/binfmt.d was not met
├─ ConditionDirectoryNotEmpty=|/usr/lib/binfmt.d was not met
├─ ConditionDirectoryNotEmpty=|/usr/local/lib/binfmt.d was not met
├─ ConditionDirectoryNotEmpty=|/etc/binfmt.d was not met
└─ ConditionDirectoryNotEmpty=|/run/binfmt.d was not met
Docs: man:systemd-binfmt.service(8)
man:binfmt.d(5)
https://docs.kernel.org/admin-guide/binfmt-misc.html
https://www.freedesktop.org/wiki/Software/systemd/APIFileSystems

Sep 13 16:54:25 Manjaro systemd[1]: Set Up Additional Binary Formats was skipped because no trigger condition checks were met.
Sep 13 16:54:46 Manjaro systemd[1]: Set Up Additional Binary Formats was skipped because no trigger condition checks were met.
Sep 13 16:59:15 Manjaro systemd[1]: Set Up Additional Binary Formats was skipped because no trigger condition checks were met.
Sep 13 17:01:51 Manjaro systemd[1]: Set Up Additional Binary Formats was skipped because no trigger condition checks were met.
```

I restarted the Vm but the problem persists. 
It makes sense that podman cannot set up the filesystem because the binaries for arm64 were not loaded. The next question is why not and how I can start systemd.binfmt.

nullx1337, 2023-09-13:
Ok i got it. 
i had to remove the qemu-user-static package with
```sh
pacman -R -dd qemu-user-static
```
and then i had to installe the qemu-user-static-binfmt package
```sh
pacman -S qemu-user-static-binfmt
```

after this i could restart systemd-binfmt.service
```sh 
sudo systemctl start systemd-binfmt.service

sudo systemctl status systemd-binfmt.service 
```

The rootfs is building now, thanks for the fast help @pythops <3

pythops, 2023-09-13:
you're welcome buddy, glad that it's fixed :)

closed 2023-09-13 by pythops