pythops/jetson-image #102: configfs not mounting under /sys/kernel/config during boot

SirMaximilian, 2023-11-07:
configfs is not mounting at boot causing the script `/opt/nvidia/l4t-usb-device-mode/nv-l4t-usb-device-mode-start.sh` to fail with `mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget’: No such file or directory`. Without the device mode the USB serial, network, and mass storage aren't starting. I'm able to mount configfs once the system boots with `mount -t configfs none /sys/kernel/config`

I'm also experiencing issues with the rootfs not expanding and dns not resolving as well.

Ive gotten Ubuntu 22.04 to build by not installing python-is-python2 and removing the python2 jetson-gpio lib with
`RUN rm /build/Linux_for_Tegra/tools/python-jetson-gpio_*_arm64.deb` after extracting the BSP

pythops, 2023-11-07:
I'm aware of those issues. Fixes are coming this week or next week. Meanwhile check the open issues, you'll find the fixes there

SirMaximilian, 2023-11-20:
any updates? could you share what you think is causing the mounting issue?

pythops, 2023-11-23:
For the DNS, you can fix it this way
```
$ sudo echo "nameserver 8.8.8.8" | tee > /etc/resolv.conf
$ sudo chattr +i /etc/resolv.conf
```

For the rootfs not expanding, can you share what makes you think it's not expanding ? the output of `lsblk -f` and `df -h` commands for instance

About the mounting issues, I never had that before. What board are you using ? 

> Ive gotten Ubuntu 22.04 to build by not installing python-is-python2 and removing the python2 jetson-gpio lib with
RUN rm /build/Linux_for_Tegra/tools/python-jetson-gpio_*_arm64.deb after extracting the BSP

Thanks for the tip.

pythops, 2023-11-24:
I confirm the issue with the resize. if you reboot then it will work. 
```
$ sudo systemctl status resizerootfs.service
```
I'll fix it in #103

pythops, 2023-11-24:
Fixed here #103

pythops, 2023-11-24:
For the mount issues, I can't reproduce the issue. the configfs is mount correctly
```
jetson@localhost:~$ mount
/dev/mmcblk0p1 on / type ext4 (rw,relatime,data=ordered)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
none on /dev type devtmpfs (rw,relatime,size=1784188k,nr_inodes=446047,mode=755)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=405916k,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/hugetlb type cgroup (rw,nosuid,nodev,noexec,relatime,hugetlb)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/debug type cgroup (rw,nosuid,nodev,noexec,relatime,debug)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=405912k,mode=700,uid=1000,gid=1000)
```

SirMaximilian, 2024-01-24:
Sorry for the late reply but this issue still persists when building with ubuntu 22.04. I've gotten it to build with 20.04 and the configfs mounts correctly.

pythops, 2024-05-22:
@SirMaximilian 
I have updated the l4t 35 to the latest available version.
I have tested with jetson orin nano and it seems to be working with ubuntu 22.04
```
root@localhost:~# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:  Ubuntu 22.04.4 LTS
Release: 22.04
Codename: jammy
root@localhost:~# mount | grep configfs
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
root@localhost:~# cat /etc/nv_tegra_release
# R35 (release), REVISION: 5.0 
```

pythops, 2024-05-27:
I am gonna close this one for the moment. feel free to comment here if you still have the issue

closed 2024-05-27 by pythops