pythops/jetson-image #109: Error: Unrecognized module SKU

raj-khare, 2023-12-25:
I'm trying to create jetson.img for Jetson AGX orin, but im running into the following err:

```
god@ubuntu:~/jetson-image$ just build-jetson-image -b jetson-agx-orin -d SD
STEP 1/21: FROM ubuntu:jammy
STEP 2/21: LABEL org.opencontainers.image.authors="Badr @pythops"
--> Using cache 5bc95df886d4300f820e26c9ce093019125d607f929d0619e987e6679eddb0bf
--> 5bc95df886d
STEP 3/21: ARG BSP=https://developer.download.nvidia.com/embedded/L4T/r35_Release_v4.1/release/Jetson_Linux_R35.4.1_aarch64.tbz2
--> Using cache 62d410647479eed86e0e391ea388994113a9384e0df7c85e88a2619b2157911d
--> 62d41064747
STEP 4/21: RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio         binutils         ssh         python-is-python3         dosfstools         python3-yaml         build-essential         lz4
--> Using cache 37f2b19622ffb531f5e12fdaa6b19a5959fa782b30132dd7c2b568f7f0cd2676
--> 37f2b19622f
STEP 5/21: RUN mkdir /build &&     wget -qO- $BSP | tar -jxpf - -C /build
--> Using cache fca56aa7454bc7b37c722c5b2def88064ce705371e4f3c2649a1bf7727110e6c
--> fca56aa7454
STEP 6/21: COPY rootfs/ /build/Linux_for_Tegra/rootfs/
--> Using cache e9c719c402d4d068046e4f5a85a9577d08feb381a203df0a6af85cfd691b4097
--> e9c719c402d
STEP 7/21: RUN chmod 4755 /build/Linux_for_Tegra/rootfs/usr/bin/sudo
--> Using cache 8eba30481e7d8d8df760d530af3170658d5646c302445fc0255c3bc54f05db3e
--> 8eba30481e7
STEP 8/21: COPY patches /patches
--> Using cache 97efe78ad28addde2db2573fd26ccd60e472b651670c494f15e446a7b32960cc
--> 97efe78ad28
STEP 9/21: RUN patch /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh < /patches/nv-apply-debs.diff &&     patch /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh < /patches/jetson-disk-image-creator.diff
--> Using cache 25067d9b9257d576b3551fd99ea7b24f210789324f9e7ac387ece072a9a5d9c5
--> 25067d9b925
STEP 10/21: RUN rm -rf /build/Linux_for_Tegra/nv_tegra/l4t_deb_packages/nvidia-l4t-nvpmodel-gui-tools*.deb &&     rm -rf /build/Linux_for_Tegra/nv_tegra/l4t_deb_packages/nvidia-l4t-jetsonpower-gui-tools*.deb
--> Using cache 36f9d2c764e66be4af3443b8c289624dba96d58fc35acbc9becaf250ce99b50b
--> 36f9d2c764e
STEP 11/21: WORKDIR /build/Linux_for_Tegra/
--> Using cache 2545a8f05b37b06cc0678ffd66cd0fbdb734ec42b6f20c2adbd16d28381ff1b2
--> 2545a8f05b3
STEP 12/21: RUN ./apply_binaries.sh
--> Using cache 3d89c949ee0c734841c9ecde47e96a9afb1405ae8f6c5d21ddf69899dcbc040b
--> 3d89c949ee0
STEP 13/21: RUN chroot /build/Linux_for_Tegra/rootfs/ /bin/bash -c "chown -R jetson:jetson /home/jetson && chmod 1777 /tmp"
--> Using cache 230ffe120c4f153a08c54ebcff2b97288e8d1288f75a9eb438884c43accfac67
--> 230ffe120c4
STEP 14/21: WORKDIR /tmp/pythops
--> Using cache 90fd3052b654ea702415b937bbcc9988b4ccb3e327203fe4b0b4aba9ef12eed8
--> 90fd3052b65
STEP 15/21: RUN gzip -d -c /build/Linux_for_Tegra/bootloader/l4t_initrd.img > initrd
--> Using cache c56450582d2b3fbd7f210b2ed4af0b4c911d665ccfa07b3a36c931cca1eeeab9
--> c56450582d2
STEP 16/21: WORKDIR /tmp/pythops/lab
--> Using cache 78345749f2456d45e39c9fc9611030c8e46e30e2c905d8386fcaa47a66dc5db0
--> 78345749f24
STEP 17/21: RUN cpio -iv < ../initrd &&     echo "nameserver 127.0.0.53" > etc/resolv.conf &&     find . | cpio --create --format="newc" > ../new_initrd
--> Using cache b9f73dc68aa64838105d2ffc1a45bbd851ac8036fda3a166785f1862562fd87d
--> b9f73dc68aa
STEP 18/21: WORKDIR /tmp/pythops
--> Using cache cb01475b9c1bf52342ca7406d82851c8220ebf649985576ea101342bec4cc619
--> cb01475b9c1
STEP 19/21: RUN gzip new_initrd && mv new_initrd.gz /build/Linux_for_Tegra/bootloader/l4t_initrd.img
--> Using cache e3134ddd39d03b78bcaa03fa19bc5b3f0f9e413af1dd80d033517ca74d30d9f0
--> e3134ddd39d
STEP 20/21: COPY scripts/create-jetson-image.sh /usr/local/bin/
--> aeb8c0163e7
STEP 21/21: WORKDIR /build/Linux_for_Tegra/tools
COMMIT jetson-build-image-l4t35
--> a92c3169114
Successfully tagged localhost/jetson-build-image-l4t35:latest
a92c31691143b95a120b3471d4d8963a3cbc63fe3ab7458d770a9b8ab79a4d71
******* board: jetson-agx-orin 
******* device: SD 
Creating image for Jetson agx orin board 
********************************************
     Jetson Disk Image Creation Tool     
********************************************
jetson-disk-image-creator.sh - creating signed images
/build/Linux_for_Tegra /build/Linux_for_Tegra/tools
###############################################################################
# L4T BSP Information:
# R35 , REVISION: 4.1
# User release: 0.0
###############################################################################
Board ID(3701) version() sku() revision()
Chip SKU(00:00:00:D0) ramcode() fuselevel(fuselevel_production) board_FAB()
Error: Unrecognized module SKU 
cp: cannot stat 'jetson.img': No such file or directory
[OK]
```

raj-khare, 2023-12-25:
I've added the following to `create-jetson-image.sh`

```
jetson-agx-orin)
    printf "Creating image for Jetson agx orin board \n"
    sudo ./jetson-disk-image-creator.sh -o jetson.img -b jetson-agx-orin-devkit -d "$JETSON_DEVICE"
    cp jetson.img /jetson/
    printf "[OK]\n"
    ;;
```

pythops, 2023-12-25:
The AGX orin is not supported yet.  I had issues creating the disk image for it following the same logic as the other boards.  Maybe with L4T 36.X  it can be fixed but I did not try yet

pythops, 2023-12-25:
I can keep this issue to open to track the support for AGX orin Board

raj-khare, 2023-12-25:
got it. how can I try 36.X in the meantime?

pythops, 2023-12-25:
You can follow the instructions here for example 
https://github.com/pythops/jetson-image/blob/master/Containerfile.image.l4t35

and change the line number 6 to download the l4t 36.X 
https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v2.0/release/jetson_linux_r36.2.0_aarch64.tbz2

and then if somethings breaks, sometimes hours of debugging might be required :wink: 

hope will it works for you :)

Good luck

pythops, 2024-01-14:
Gonna close this one as it points to two different open issues. Progress will be on those issues #113 and #108

closed 2024-01-14 by pythops