pythops/jetson-image #114: cp: cannot create regular file '/jetson/jetson.img'

Dviros, 2024-01-19:
Hi,
Thank you very much for this project.
I'm attempting to build an image for Jetson Nano B1 with Ubuntu 22.
I'm using macOS and these are the commands I've been using:
- just build-jetson-rootfs 22.04
- just build-jetson-image -b jetson-nano -r 300 -d SD

The output I get is super strange, especially because the path "/jetson/jetson.img" exists only in the  create-jetson-image.sh script, but also mentioned in a file that I cannot find - jetson-disk-image-creator.sh. I need help to proceed in here please :)

Thanks again,
D

```
sudo just build-jetson-image -b jetson-nano -r 300 -d SD
STEP 1/21: FROM ubuntu:jammy
STEP 2/21: LABEL org.opencontainers.image.authors="Badr @pythops"
--> Using cache dd9e218ec9bf57499f89aceae1e9df51ebbddec1e5d3d83b6bfa906b05b8e954
--> dd9e218ec9bf
STEP 3/21: ARG BSP=https://developer.download.nvidia.com/embedded/L4T/r32_Release_v7.4/T210/Jetson-210_Linux_R32.7.4_aarch64.tbz2
--> Using cache 77930c199b8f969d02e59f1d1e0a42149d383d75474c2eb288e2ad6aa1506c47
--> 77930c199b8f
STEP 4/21: RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio
--> Using cache 18cf7b295aba88fd1bd24d01e89b9ff64222b5208d41adef1469e75f35c7c38f
--> 18cf7b295aba
STEP 5/21: RUN mkdir /build &&     wget -qO- $BSP | tar -jxpf - -C /build
--> Using cache 6d4f1cc6538c4f87f46c3e6be8de304640e856c9071b2d923f8bc9c96d544cd7
--> 6d4f1cc6538c
STEP 6/21: COPY rootfs/ /build/Linux_for_Tegra/rootfs/
--> 9212113afa61
STEP 7/21: RUN chmod 4755 /build/Linux_for_Tegra/rootfs/usr/bin/sudo
--> c2d97962b9a7
STEP 8/21: COPY patches /patches
--> c894c3f18221
STEP 9/21: RUN patch /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh < /patches/nv-apply-debs.diff &&     patch /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh < /patches/jetson-disk-image-creator.diff
patching file /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
patching file /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh
--> 0b00756947ad
STEP 10/21: RUN rm -rf /build/Linux_for_Tegra/tools/python-jetson-gpio_*_arm64.deb
--> e6d768ab6e99
STEP 11/21: WORKDIR /build/Linux_for_Tegra/
--> 2b69830929b4
STEP 12/21: RUN ./apply_binaries.sh
Using rootfs directory of: /build/Linux_for_Tegra/rootfs
Installing extlinux.conf into /boot/extlinux in target rootfs
/build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
Root file system directory is /build/Linux_for_Tegra/rootfs
Copying public debian packages to rootfs
Start L4T BSP package installation
QEMU binary is not available, looking for QEMU from host system
Found /usr/bin/qemu-aarch64-static
Installing QEMU binary in rootfs
/build/Linux_for_Tegra/rootfs /build/Linux_for_Tegra
Installing BSP Debian packages in /build/Linux_for_Tegra/rootfs
Selecting previously unselected package nvidia-l4t-core.
(Reading database ... 49741 files and directories currently installed.)
Preparing to unpack .../nvidia-l4t-core_32.7.4-20230608212426_arm64.deb ...
Pre-installing... skip compatibility checking.
Unpacking nvidia-l4t-core (32.7.4-20230608212426) ...
Setting up nvidia-l4t-core (32.7.4-20230608212426) ...
Selecting previously unselected package jetson-gpio-common.
(Reading database ... 49787 files and directories currently installed.)
Preparing to unpack .../jetson-gpio-common_2.0.17_arm64.deb ...
Unpacking jetson-gpio-common (2.0.17) ...
Selecting previously unselected package python3-jetson-gpio.
Preparing to unpack .../python3-jetson-gpio_2.0.17_arm64.deb ...
Unpacking python3-jetson-gpio (2.0.17) ...
Selecting previously unselected package nvidia-l4t-3d-core.
Preparing to unpack .../nvidia-l4t-3d-core_32.7.4-20230608212426_arm64.deb ...
Unpacking nvidia-l4t-3d-core (32.7.4-20230608212426) ...
Selecting previously unselected package nvidia-l4t-apt-source.
[... 572 lines cut ...]
127488 bytes (127 kB, 124 KiB) copied, 0.0033863 s, 37.6 MB/s
jetson-disk-image-creator.sh - writing /build/Linux_for_Tegra/bootloader/system.img.raw
5304320+0 records in
5304320+0 records out
2715811840 bytes (2.7 GB, 2.5 GiB) copied, 30.8159 s, 88.1 MB/s
del devmap : loop0p1
del devmap : loop0p10
del devmap : loop0p11
del devmap : loop0p12
del devmap : loop0p13
del devmap : loop0p14
del devmap : loop0p2
del devmap : loop0p3
del devmap : loop0p4
del devmap : loop0p5
del devmap : loop0p6
del devmap : loop0p7
del devmap : loop0p8
del devmap : loop0p9
********************************************
   Jetson Disk Image Creation Complete
********************************************
cp: cannot create regular file '/jetson/jetson.img': Operation not permitted
[OK]
```

pythops, 2024-01-19:
Seems to be a permission issue. The image has been generated but can not be copied to your host fs.
If you can run the generation scripts on a Linux VM that would solve the issue I think.

Dviros, 2024-01-19:
Bummer :( any way to trace to the exact copy command?
I'll try to run it in a VM

pythops, 2024-01-19:
In the `create-jetson-image.sh`, there is `cp jetson.img /jetson/` which copy the image from the container to the host machine.

Dviros, 2024-01-19:
Yep, modified it to point to a controlled folder, but the original path still reflects in the logs. It seems that value appears in `jetson-disk-image-creator.sh`, which is different from the one that exists in the scripts folder.

Dviros, 2024-02-08:
Solved,
This was achieved by running
- `docker cp $(docker ps -qf "name=gitated_gould"):/build/Linux_for_Tegra/tools/jetson.img ~/jetson.img`
Maybe adding a switch case to check the OS and run it to extract the img from the docker?
Thanks again for your hard work.

pythops, 2024-02-11:
Glad that you solved it. I'll add a note to the Readme for macos users

closed 2024-02-11 by pythops

egeakman, 2024-03-05:
I just did ``touch jetson.img`` before building the Jetson image and it worked.