pythops/jetson-image #115: Password is not working

ray-inb, 2024-02-09:
I have tried both your per-build image and step-wise build image, it stucked at password login. I used 'jetson' but not working. Here is partial build log details                                                               
                                                                                                                                                                    
[    6.974451] pva_iommu_context_dev 16000000.pva0:pva0_niso1_ctx7: initialized)
[    6.985822] pva 16000000.pva0: Adding to iommu group 48                      
[    6.991598] nvdla 15880000.nvdla0: Adding to iommu group 49                  
[    6.997665] nvdla 158c0000.nvdla1: Adding to iommu group 50                  
[    7.003828] tegra-gpcdma 2600000.gpcdma: Adding to iommu group 2             
[    7.010997] tegra-gpcdma 2600000.gpcdma: GPC DMA driver register 31 channels 
[    7.018530] tegra-fuse-burn efuse-burn: shutdown limit check disabled        
[    7.025149] tegra-fuse-burn efuse-burn: Fuse burn driver initialized         
[    7.031883] serial-tegra 3100000.serial: Adding to iommu group 2             
[    7.038481] 3100000.serial: ttyTHS0 at MMIO 0x3100000 (irq = 17, base_baud =T
[    7.048285] serial-tegra 3130000.serial: Adding to iommu group 2             
[    7.054793] 3130000.serial: ttyTHS3 at MMIO 0x3130000 (irq = 63, base_baud =T
[    7.064560] serial-tegra 3140000.serial: Adding to iommu group 2             
[    7.071124] 3140000.serial: ttyTHS4 at MMIO 0x3140000 (irq = 64, base_baud =T
[    7.097718] tegra-xusb 3610000.xhci: Firmware timestamp: 2021-12-01 05:27:10e
[    7.107463] tegra-xusb 3610000.xhci: xHCI Host Controller                    
[    7.113036] tegra-xusb 3610000.xhci: new USB bus registered, assigned bus nu1
[    7.121716] tegra-xusb 3610000.xhci: hcc params 0x0180ff05 hci version 0x1200
[    7.131334] tegra-xusb 3610000.xhci: irq 230, io mem 0x03610000              
[    7.137789] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, b0
[    7.146284] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber1
[    7.153696] usb usb1: Product: xHCI Host Controller                          
[    7.158702] usb usb1: Manufacturer: Linux 5.10.120-tegra xhci-hcd            
[    7.164962] usb usb1: SerialNumber: 3610000.xhci                             
[    7.170493] hub 1-0:1.0: USB hub found                                       
[    7.174375] hub 1-0:1.0: 4 ports detected                                    
[    7.178747] tegra-xusb 3610000.xhci: xHCI Host Controller                    
[    7.184300] tegra-xusb 3610000.xhci: new USB bus registered, assigned bus nu2
[    7.192081] tegra-xusb 3610000.xhci: Host supports USB 3.1 Enhanced SuperSped
[    7.199587] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, b0
[    7.208093] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber1
[    7.215520] usb usb2: Product: xHCI Host Controller                          
[    7.220522] usb usb2: Manufacturer: Linux 5.10.120-tegra xhci-hcd            
[    7.226777] usb usb2: SerialNumber: 3610000.xhci                             
[    7.232133] hub 2-0:1.0: USB hub found                                       
[    7.236001] hub 2-0:1.0: 4 ports detected                                    
[    7.263299] i2c i2c-2: Added multiplexed i2c bus 9                           
[    7.269007] i2c i2c-2: Added multiplexed i2c bus 10                          
[    7.274023] i2c-mux-gpio cam_i2cmux: 2 port mux on 3180000.i2c adapter       
[    7.281010] tegra23x-oc-event soctherm-oc-event: Finished tegra23x overcurreg
[    7.302858] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequz
[    7.315538] cpufreq: cpufreq_online: CPU4: Running at unlisted initial frequz
[    7.327442] tegra194-cpufreq e000000.ccplex: probed with ICC, cl:3           
[    7.335377] tegra194-pcie 14100000.pcie: Using GICv2m MSI allocator          
[    7.336776] sdhci-tegra 3400000.sdhci: Got CD GPIO                           
[    7.342890] tegra194-pcie 14100000.pcie: host bridge /pcie@14100000 ranges:  
[    7.353943] tegra194-pcie 14100000.pcie:       IO 0x0030100000..0x00301fffff0
[    7.356053] irq: IRQ294: trimming hierarchy from :interrupt-controller@f40001
[    7.362630] tegra194-pcie 14100000.pcie:      MEM 0x20a8000000..0x20afffffff0
[    7.370299] sdhci-tegra 3400000.sdhci: wakeup init done, cdirq 294           
[    7.378930] tegra194-pcie 14100000.pcie:      MEM 0x2080000000..0x20a7ffffff0
[    7.453560] usb 1-2: new high-speed USB device number 2 using tegra-xusb     
[    7.485745] tegra194-pcie 14100000.pcie: Link up                             
[    7.492180] tegra194-pcie 14100000.pcie: PCI host bridge to bus 0001:00      
[    7.499001] pci_bus 0001:00: root bus resource [bus 00-ff]                   
[    7.504635] pci_bus 0001:00: root bus resource [io  0x0000-0xfffff] (bus add)
[    7.514471] pci_bus 0001:00: root bus resource [mem 0x20a8000000-0x20affffff)
[    7.525298] pci_bus 0001:00: root bus resource [mem 0x2080000000-0x20a7fffff]
[    7.533232] pci 0001:00:00.0: [10de:229e] type 01 class 0x060400             
[    7.539575] pci 0001:00:00.0: PME# supported from D0 D3hot                   
[    7.550854] pci 0001:01:00.0: [10ec:c822] type 00 class 0x028000             
[    7.557190] pci 0001:01:00.0: reg 0x10: [io  0x0000-0x00ff]                  
[    7.563123] pci 0001:01:00.0: reg 0x18: [mem 0x20a8000000-0x20a800ffff 64bit]
[    7.571591] pci 0001:01:00.0: supports D1 D2                                 
[    7.575979] pci 0001:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold      
[    7.588107] pci 0001:00:00.0: BAR 14: assigned [mem 0x20a8000000-0x20a80ffff]
[    7.595530] pci 0001:00:00.0: BAR 13: assigned [io  0x1000-0x1fff]           
[    7.601880] pci 0001:01:00.0: BAR 2: assigned [mem 0x20a8000000-0x20a800ffff]
[... 216 lines cut ...]
[   15.011733]                                                                  
[   15.098682] random: crng init done                                           
[   15.102228] random: 18 urandom warning(s) missed due to ratelimiting         
[   15.197107] systemd-journald[247]: Received client request to flush runtime .
[   15.388822] systemd-journald[247]: File /var/log/journal/2c41f9ba60475501bc6.
[   16.157584] nvidia: loading out-of-tree module taints kernel.                
[   16.174898] imx219 9-0010: imx219_board_setup: error during i2c read probe ()
[   16.187744] imx219 9-0010: board setup failed                                
[   16.192333] imx219: probe of 9-0010 failed with error -121                   
[   16.209307] imx219 10-0010: imx219_board_setup: error during i2c read probe )
[   16.219466] fusb301 1-0025: toggle_time(0) is not updated                    
[   16.228811] imx219 10-0010: board setup failed                               
[   16.235104] imx219: probe of 10-0010 failed with error -121                  
[   16.364887] platform regulatory.0: Direct firmware load for regulatory.db fa2
[   16.373786] platform regulatory.0: Falling back to sysfs fallback for: regulb
[   16.824214] using random self ethernet address                               
[   16.829395] using random host ethernet address                               
[   17.481375] using random self ethernet address                               
[   17.486307] using random host ethernet address                               
                                                                                
Ubuntu 20.04.6 LTS localhost ttyTCU0                                            
                                                                                
localhost login: 

It stucked in here nither accepting the password nor showing any error or any further output. What to do ?

ray-inb, 2024-02-09:
I am using Jetson Orin Nano Devkit 4GB version.

pythops, 2024-02-09:
1. where are those logs from ? which build part ?
2. have you done any modification to the scripts ? 
3. When you tried to build the image, did it go without any error ?

ray-inb, 2024-02-10:
I observed this log in UART terminal. I did not do any modification. Yes, I can build the image without any error. 

Initially I used (a I have jetson orin nano devkit)
sudo just build-jetson-image -b jetson-orin-nano-devkit -d SD
but then I got these error 
The supported boards are:
- jetson-nano
- jetson-nano-2gb
- jetson-orin-nano
- jetson-agx-xavier
- jetson-xavier-nx

then I used only 'sudo just build-jetson-image -b jetson-orin-nano -d SD' and it created the image without any error.

ray-inb, 2024-02-10:
I have added display , keyboard, mouse with the board but display was all black ()though as I understand the image does not have any UI ) and typed the password but did not work either.

pythops, 2024-02-11:
Strange ! 
you're using `jetson` as username and `jetson` as password right ?