pythops/jetson-image #117: Error on compiling ubuntu 22.04

nlnen, 2024-02-11:
`Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
Segmentation fault (core dumped)
Segmentation fault (core dumped)
dpkg: error processing package libc-bin (--configure):
 installed libc-bin package post-installation script subprocess returned error exit status 139
Processing triggers for ca-certificates (20230311ubuntu0.22.04.1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)`

It seems that there is some memory wrong allocated due to Segmentation fault.
Compiling ubuntu20.04 works fine without any problems

pythops, 2024-02-12:
what are the commands you have ran to get this ? what is your OS  ?

nlnen, 2024-02-12:
My System ist fedora 39 Workstation
The command is "just build-jetson-rootfs 22.04"

nlnen, 2024-02-12:
What I originally wanted was a working Ubuntu 22.04 image (for jetson nano). But since this was not to be found on the releases page, I wanted to compile it myself. However, only version 20.04 worked fine.

pythops, 2024-02-12:
I'll take a look at it

nlnen, 2024-02-12:
In case it worked for you, could you publish the Ubuntu 22.04 images in the releases?

pythops, 2024-02-12:
Sure

pythops, 2024-02-13:
I could not reproduce it actually.
Here are some tips you can try:
1. try to get the latest version of podman
2. If you have updated the kernel, you might need to restart your computer

I'll try to publish 22.04 images this week or next week

nekiwo, 2024-08-09:
I had a similar issue with exact same output when building using podman 4.9.5 on Ubuntu 24.04. The solution was to add the following lines to `Containerfile.rootfs.22_04` as outlined in this [StackOverflow answer](https://stackoverflow.com/questions/78105004/docker-build-fails-because-unable-to-install-libc-bin).

```
RUN rm /var/lib/dpkg/info/libc-bin.*
RUN apt-get clean
RUN apt-get update
RUN apt-get install libc-bin
```

TWTom041, 2024-08-12:
Since no Ubuntu 22.04 image release is obtainable in the release page, I think this should help. [Ubuntu 22.04 jetson nano image](https://github.com/TWTom041/jetson-image/releases/tag/2024-08-04)

pythops, 2024-08-12:
Send PR ;)

TWTom041, 2024-08-12:
@pythops Does it merge releases? I don't remember a pr will merge releases. Since most of the script I changed is some hacky modifications, I don't recommend merging my branch.

pythops, 2024-08-12:
What I mean, is you're welcome to send changes to build Ubuntu 22.04 images on the CI, if you want ofc :)

ijiki16, 2024-11-07:
I have error compiling Ubuntu 22.04 on Fedora 41 ( Ubuntu 20.04 works normally)

package: `libc-bin` 

```bash
Setting up libgtk-3-0:arm64 (3.24.33-1ubuntu2.2) ...
Setting up libgtk-3-bin (3.24.33-1ubuntu2.2) ...
Setting up humanity-icon-theme (0.6.16) ...
Setting up ubuntu-mono (20.10-0ubuntu2) ...
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
dpkg: error processing package libc-bin (--configure):
 installed libc-bin package post-installation script subprocess returned error exit status 139
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.8+dfsg-1ubuntu0.3) ...
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)
Error: building at STEP "RUN apt install -y         libgles2         libpangoft2-1.0-0         libxkbcommon0         libwayland-egl1         libwayland-cursor0         libunwind8         libasound2         libpixman-1-0         libjpeg-turbo8         libinput10         libcairo2         device-tree-compiler         iso-codes         libncursesw5         libdrm-common         libdrm2         libegl-mesa0         libegl1         libegl1-mesa         libgtk-3-0         libgstreamer1.0-0         libgstreamer-plugins-bad1.0-0         i2c-tools         bridge-utils         libffi7         libvulkan1         libssl1.1 &&         apt install -y wget cpio busybox &&         wget http://launchpadlibrarian.net/354032550/libffi6_3.2.1-8_arm64.deb &&         dpkg -i libffi6_3.2.1-8_arm64.deb": while running runtime: exit status 100
```

<details>
<summary>Full log after running: `just build-jetson-rootfs 22.04` </summary>
<br>
```bash
Building base rootfs
STEP 1/13: FROM ubuntu:jammy
STEP 2/13: LABEL org.opencontainers.image.authors="Badr @pythops"
STEP 3/13: RUN echo "deb http://ports.ubuntu.com/ubuntu-ports/ focal main" >> /etc/apt/sources.list  &&     apt update

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease [270 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [128 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [127 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [129 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports focal InRelease [265 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages [17.2 MB]
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse arm64 Packages [224 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports jammy/restricted arm64 Packages [24.2 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 Packages [1758 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [2346 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Packages [1400 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse arm64 Packages [29.5 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted arm64 Packages [2597 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main arm64 Packages [80.9 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe arm64 Packages [31.8 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [2077 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted arm64 Packages [2528 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse arm64 Packages [24.2 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packages [1118 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 Packages [1234 kB]
Fetched 33.6 MB in 13s (2601 kB/s)
Reading package lists...
Building dependency tree...
Reading state information...
All packages are up to date.
STEP 4/13: RUN apt install -y         libgles2         libpangoft2-1.0-0         libxkbcommon0         libwayland-egl1         libwayland-cursor0         libunwind8         libasound2         libpixman-1-0         libjpeg-turbo8         libinput10         libcairo2         device-tree-compiler         iso-codes         libncursesw5         libdrm-common         libdrm2         libegl-mesa0         libegl1         libegl1-mesa         libgtk-3-0         libgstreamer1.0-0         libgstreamer-plugins-bad1.0-0         i2c-tools         bridge-utils         libffi7         libvulkan1         libssl1.1 &&         apt install -y wget cpio busybox &&         wget http://launchpadlibrarian.net/354032550/libffi6_3.2.1-8_arm64.deb &&         dpkg -i libffi6_3.2.1-8_arm64.deb

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  adwaita-icon-theme alsa-topology-conf alsa-ucm-conf at-spi2-core dbus
  dbus-user-session dconf-gsettings-backend dconf-service dmsetup fontconfig
  fontconfig-config fonts-dejavu-core gir1.2-glib-2.0
[... 1131 lines cut ...]
Created symlink /etc/systemd/system/multi-user.target.wants/networkd-dispatcher.service → /lib/systemd/system/networkd-dispatcher.service.
Setting up gsettings-desktop-schemas (42.0-1ubuntu1) ...
Setting up at-spi2-core (2.44.0-3) ...
Setting up adwaita-icon-theme (41.0-1ubuntu1) ...
[... cut ...]       libdrm2         libegl-mesa0         libegl1         libegl1-mesa         libgtk-3-0         libgstreamer1.0-0         libgstreamer-plugins-bad1.0-0         i2c-tools         bridge-utils         libffi7         libvulkan1         libssl1.1 &&         apt install -y wget cpio busybox &&         wget http://launchpadlibrarian.net/354032550/libffi6_3.2.1-8_arm64.deb &&         dpkg -i libffi6_3.2.1-8_arm64.deb": while running runtime: exit status 100
```
</details>

Can we reopen this issue?

pythops, 2024-11-07:
Sure,

Can you write down the commands you run and eventual changes you made ?

ijiki16, 2024-11-08:
@pythops  I just run `just build-jetson-rootfs 22.04`.
I also tested 22.04 builds from @TWTom041 for jetson nano 2gb & it didn't worked.
This is Searial/UART logs from that image: [jetson-nano-2gb-ubuntu-22.04.log](https://github.com/user-attachments/files/17677380/jetson-nano-2gb-ubuntu-22.04.log)

pythops, 2024-11-08:
Did you make any changes to the rootfs (add new packages) ? Please detail here the steps to reproduce the issue ?

ijiki16, 2024-11-08:
No I didn't, I am just trying to build from repo without any changes.