pythops/jetson-image #118: Certificated not valid and failed sync by default date doesn't working correctly

mrcmunir, 2024-02-20:
```
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP: 23.33.143.169 443]
Hit:6 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Err:7 https://repo.download.nvidia.com/jetson/t210 r32.7 Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP: 23.33.143.169 443]
Get:8 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Reading package lists... Done     
E: The repository 'https://repo.download.nvidia.com/jetson/common r32.7 Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'https://repo.download.nvidia.com/jetson/t210 r32.7 Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Release file for http://ports.ubuntu.com/ubuntu-ports/dists/focal-backports/InRelease is not valid yet (invalid for another 230d 6h 25min 31s). Updates for this repository will not be applied.
E: Release file for http://ports.ubuntu.com/ubuntu-ports/dists/focal-updates/InRelease is not valid yet (invalid for another 329d 16h 11min 56s). Updates for this repository will not be applied.
E: Release file for http://ports.ubuntu.com/ubuntu-ports/dists/focal-security/InRelease is not valid yet (invalid for another 329d 15h 44min 43s). Updates for this repository will not be applied.
``

Seems it's related due the service systemd-timesync not enabled by default so ntp sync doesn't working by default and cause apt certificate and sync update will fail .

pythops, 2024-02-20:
Yes, this is because of time. 
try to setup the ntp for the system as follows
```
sudo timedatectl set-ntp true
```
That should fix it

closed 2024-02-20 by pythops

mrcmunir, 2024-02-20:
Thanks i'm solved with enabled service systemd-timesync start ON by default for some reason it's dead :p

pythops, 2024-02-20:
Maybe I should enable that service by default when building the image. 
I'll create an issue related to it

pythops, 2024-02-20:
https://github.com/pythops/jetson-image/issues/119