pythops/jetson-image #119: Enable systemd-timesyncd svc

pythops, 2024-02-20:


pgluz, 2024-04-06:
Have you found any solution for this issue? In my case it's loaded but inactive due to condition:
```
start condition failed at Sat 2000-01-01 01:32:37 UTC; 1min 15s ago
 - ConditionVirtualization=!container was not met
```

which is weird because I'm running it on bare Jetson Nano 4 GB. Ubuntu 22.04 was made with your tutorial and "just" tool and it seems to work fine otherwise, but because of this NTP issue it's impossible to use apt.

I also noticed that timedatectl status prints this:
`[273.840637] proc: unrecognized mount option "hidepid=invisible" or missing value`

As [I found in the Internet](https://github.com/systemd/systemd/issues/16896), `hidepid=invisible` should not be used in kernels <5.8. Not sure how to fix it and if it has anything to do with this bug, but I think it's worth to mention.

pythops, 2024-05-22:
Done

closed 2024-05-22 by pythops