pythops/jetson-image #121: Bug: `sd-bus call: Transport endpoint is not connected`

jwdinius, 2024-03-31:
After building the rootfs using `just build-jetson-rootfs 22.04`, I tried building the disk image and caught an error in the first step:

```bash
$ just build-jetson-image -b jetson-agx-xavier -d USB
STEP 1/21: FROM ubuntu:jammy
STEP 2/21: LABEL org.opencontainers.image.authors="Badr @pythops"
--> Using cache 6000c4945bc504e18b583afeb7beb8637dbd9b6a120f1a6aca80daae5a00c263
--> 6000c4945bc
STEP 3/21: ARG BSP=https://developer.download.nvidia.com/embedded/L4T/r35_Release_v4.1/release/Jetson_Linux_R35.4.1_aarch64.tbz2
--> Using cache 62ae9545687f9f470c7685c7a31a0fcf63f0c6fec6107f1ea6065362a8f2b47e
--> 62ae9545687
STEP 4/21: RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio         binutils         ssh         python-is-python3         dosfstools         python3-yaml         build-essential         lz4
error running container: error from /usr/bin/crun creating container for [/bin/sh -c apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio         binutils         ssh         python-is-python3         dosfstools         python3-yaml         build-essential         lz4]: sd-bus call: Transport endpoint is not connected
: exit status 1
Error: error building at STEP "RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio         binutils         ssh         python-is-python3         dosfstools         python3-yaml         build-essential         lz4": error while running runtime: exit status 1
```

This appears to be related to the `sudo -E ...` lines [here](https://github.com/pythops/jetson-image/blob/96f97645f98d0540609b62ef9302c6cbcc4c7096/scripts/build-jetson-image.sh#L90) and [here](https://github.com/pythops/jetson-image/blob/96f97645f98d0540609b62ef9302c6cbcc4c7096/scripts/build-jetson-image.sh#L96); see this [related issue](https://github.com/containers/buildah/issues/3887)

pythops, 2024-03-31:
Interesting one, never had that issue before, plus the CI worked :thinking:

jwdinius, 2024-03-31:
Yeah, because of the local scope of the change, the fix seems low-risk.  The settings won't persist past the call.

PR #122 (2024-03-31): unset DBUS_SESSION_BUS_ADDRESS in build-jetson-image.sh (fixes #121)
fixes #121
commit 10ff93fd (2024-03-31): unset DBUS_SESSION_BUS_ADDRESS in build-jetson-image.sh (fixes #121) (#122)
closed 2024-03-31 by pythops