pythops/jetson-image #125: Issues with Jetson nano not booting on jetson-nano-300_20.04_32.7.4

Chrescht, 2024-05-22:
Apart from the fact my Nano not booting up (blank screen), I have several questions:

- what's the 300 for in jetson-nano-300_20.04_32.7.4.img.lrz ?
- why is there no 22.04 files but the Readme talks about both versions 20.04 and 22.04 ?
- is the **default login** valid for ssh ?

Many thanks.

pythops, 2024-05-22:
> what's the 300 for in jetson-nano-300_20.04_32.7.4.img.lrz ?

It is for the revision. for A02 board it is 200, for the B01 it is 300 [link](https://www.seeedstudio.com/blog/2020/01/16/new-revision-of-jetson-nano-dev-kit-now-supports-new-jetson-nano-module/)

> why is there no 22.04 files but the Readme talks about both versions 20.04 and 22.04 ?

The Readme states that 20.04 and 22.04 are supported if  you want to download them or build them. There is another img.lz with 22.04 in the name 

> is the default login valid for ssh ?

Yes

Chrescht, 2024-05-22:
Thanks for the prompt reply.

> The Readme states that 20.04 and 22.04 are supported if you want to download them or build them. There is another img.lz with 22.04 in the name

I found no files for 22.04 among [the releases](https://github.com/pythops/jetson-image/releases).

pythops, 2024-05-22:
Yeah it is missing :thinking:  
From the CI file, it should be built. I am gonna merge soon the PR for 36.x BSP and then I will trigger the build again.
Meanwhile, you can build the image based on ubuntu 22.04, easy and straightforward.

pythops, 2024-05-22:
Gonna close this one. feel free to open an issue if you encounter any issue

closed 2024-05-22 by pythops