pythops/jetson-image #131: Flash boards without SD card, like "Jetson Xavier NX"

HamzeSol, 2024-05-28:
### Issue:
First of all, thank you for this fantastic project. The minimalist images are incredibly helpful and save a lot of space and resources.

### Problem:
Your current method suggests flashing the Jetson boards using an SD card. However, the Jetson Xavier NX does not have an SD card slot. Instead, it requires flashing the internal eMMC memory. A method or guide to flash the eMMC memory of boards like Jetson Xavier NX is needed.

Thank you for considering it!

pythops, 2024-05-28:
Sure, that's something I can add. But unfortunately I do not have those boards so I will need you to test the solution I will build, sounds good ?

HamzeSol, 2024-05-28:
Sounds great!
Many thanks, I will test the solution as soon as you add it :)

jobesu14, 2024-12-17:
I can also help on the testing if needed.

pythops, 2024-12-17:
Great, which board do you have ?

jobesu14, 2024-12-17:
The Jetson Xavier NX

nullset2, 2025-03-21:
Hi. I am trying to use your tool because NVIDIA's official support is a joke, and so far so good, I love this repo. However, I can't seem to be able to flash the image to my board.

I am using a Turing Pi V2 and it does get recognized by the official SDKmanager app from NVIDIA. I am flashing a Jetson Nano module rev 200. Neither nvme0n1p1 or mmcblk0p1 come up as devices in the filesystem. It'd be useful if some context could be provided for situations like these.