pythops/jetson-image #132: Minimum image with CUDA support

mpaszkuta, 2024-06-05:
First of all, I would like to thank you for this project. It helps me save space on the SD card and time during development. I aim to create an image with the minimum CUDA tools installed for PyTorch Python development. There is a special version of PyTorch for Jetson modules that requires CUDA. Normally, these tools are installed with JetPack. The CUDA tools can be installed after the L4T integration. I have prepared a list of the minimum packages that need to be installed, but it is not possible to add this to Containerfile.rootfs.22_04 during the rootfs build. However, this should be realized in Containerfile.rootfs.22_04 after new sources are added. It would be beneficial to add the possibility of installing selected packages from JetPack, such as CUDA, OpenCV, etc.

pythops, 2024-06-10:
Here is the PR to solve this issue https://github.com/pythops/jetson-image/pull/133
Just add the packages you want in the file l4t_packages.txt and build following the same instructions, 
I have tested it locally and it seems to be working well.
Please give it a try and let me know if this approach suits you

pythops, 2024-06-12:
I would like to merge the PR. please take a moment and give it a try. Any feedback is welcome :)

pythops, 2024-06-14:
Merged

closed 2024-06-14 by pythops