pythops/jetson-image #135: Flashing the jetson-image by flash.sh, Fails to Boot on Subsequent Attempts Description:

HamzeSol, 2024-06-20:
After flashing your jetson-image using **`flash.sh`** , it successfully boots on the first attempt. However, subsequent attempts to boot fail. The initial boot is successful, but subsequent boots result in failure.

Target board is jetson-nano-devkit and flashed by USB cable in recovery mode!

I tried two methods:
method 1: 
- copy your rootfs in the /Linux_for_tegra/rootfs 
- sudo ./apply_binaries.sh
- sudo ./flash.sh jetson-nano-devkit mmcblk0p1

method 2: 
- mount your jetson.img
- copy its partitions into /Linux_for_tegra/rootfs
- sudo ./apply_binaries.sh
- sudo ./flash.sh jetson-nano-devkit mmcblk0p1

I tried two methods but encountered same result:

Here is the boot log on serial port:

```
[0003.872] [L4T TegraBoot] (version ���'���k��/+2w���z�����@�\��) [0003.878] Processing in recovery mode [0003.882] A02 Bootrom Patch rev = 1023 [0003.885] Power-up reason: pmc por [0003.889] Established communication link with host [0004.939] NvTbootI2cWrite(): error code 0x00045100 Error while starting write transaction [0004.947] NvTbootI2cDeviceRead(): error code 0x00045001 Error while sending the offset to slave [0004.955] NvTbootI2c: Read failed for slave 0xac, offset 0x00 with error code 0x00045001 [0004.060] Enabled early print [0004.063] [L4T TegraBoot] (version ���'���k��/+2w���z�����@�\��)
[0004.069] Processing in recovery mode
[0004.073] A02 Bootrom Patch rev = 1023
[0004.076] Power-up reason: software reset
[0004.080] Established communication link with host
[0005.269] Odmdata from BCT: 0x000a4000
[0005.273] DebugPort= 0x3
[0005.275] SkipQspiOnly= 0x0
[0005.327] BoardId read from EEPROM/NCT: 3448
[0005.331] BoardID = 3448, SKU = 0x0
[0005.359] NvTbootInit_Porg: Setting GPIO_A6 for HDMI ...
[0005.364] max77620 setting MBLPD bit
[0005.367] No Battery Present
[0005.370] RamCode = 0
[0005.372] Platform has DDR4 type RAM
[0005.376] max77620 disabling SD1 Remote Sense
[0005.380] Setting DDR voltage to 1125mv
[0005.384] Serial Number of Pmic Max77663: 0x2003ec
[0005.391] Entering ramdump check
[0005.394] Get RamDumpCarveOut = 0x0
[0005.398] RamDumpCarveOut=0x0, RamDumperFlag=0xe59ff3f8
[0005.403] Last reboot was clean, booting normally!
[0005.407] Sdram initialization is successful
[0005.412] SecureOs Carveout Base=0x00000000ff800000 Size=0x00800000
[0005.418] Lp0 Carveout Base=0x00000000ff780000 Size=0x00001000
[0005.423] BpmpFw Carveout Base=0x00000000ff700000 Size=0x00080000
[0005.429] GSC1 Carveout Base=0x00000000ff600000 Size=0x00100000
[0005.435] GSC2 Carveout Base=0x00000000ff500000 Size=0x00100000
[0005.441] GSC4 Carveout Base=0x00000000ff400000 Size=0x00100000
[0005.447] GSC5 Carveout Base=0x00000000ff300000 Size=0x00100000
[0005.453] GSC3 Carveout Base=0x000000017f300000 Size=0x00d00000
[0005.469] RamDump Carveout Base=0x00000000ff280000 Size=0x00080000
[0005.475] Platform-DebugCarveout: 0
[0005.478] Downloaded BCT successfully
[0005.658] Downloaded Bootloader successfully
[0005.714] Downloaded rp1 successfully
[0005.798] MAX77620_GPIO5 configured
[0005.801] CPU power rail is up
[0005.804] CPU clock enabled
[0005.808] Performing RAM repair
[0005.811] Updating A64 Warmreset Address to 0x92c002e9
[0005.817] Enable APE clock/reset
[0005.820] Error in NvTbootGetTOSBinaryLength: 0x11 !
[0005.824] Loading Secure OS image failed.
[0005.828] Set NvDecSticky Bits
[0005.832] GSC2 address ff53fffc value c0edbbcc
[0005.838] GSC MC Settings done
[0005.842] Next binary entry address: 0x92c00258
[0005.847] BoardId: 3448
[... 613 lines cut ...]
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
Tegra210 (P3450-0000) #
```

pythops, 2024-06-20:
I can give it a try with my board this weekend

pythops, 2024-10-23:
@HamzeSol sorry for disappearing. 
Did u solve it ?

pythops, 2024-11-06:
any updates here ?

pythops, 2024-11-11:
I am closing this one as there is no update on it.

closed 2024-11-11 by pythops