pythops/jetson-image #136: Error building Jetson Nano 24.04 image

CristianEduardMihai, 2024-06-25:
Hi! I'm trying to build a ubuntu 22.04 image for the Jetson nano 4gb dev kit.
`just build-jetson-rootfs 24.04` ran without any issue, however `just build-jetson-image -b jetson-nano -d SD -l 36` encountered some dependency erros:
[logs-build.txt](https://github.com/user-attachments/files/15970917/logs-build.txt)
I haven't tested to see if the same error would come up with another version, like 22.04, but when i do it i'll update the issue.

pythops, 2024-06-25:
Jetson nano does not support ubuntu 24.04. The reason is that some nvidia libraries depend on some tools that are not in ubuntu 24.04. 
You option is to choose either ubuntu 20.04 or 22.04

closed 2024-06-25 by pythops

CristianEduardMihai, 2024-06-25:
Update: Ubuntu 22.04 seems to error out at installing cuda-toolkit
[build-logs-22-04.txt](https://github.com/user-attachments/files/15971400/build-logs-22-04.txt)
Will post an update after removing the `cuda-toolkit` from `Containerfile.image.*`

pythops, 2024-06-25:
You can not use `-l 36` for jetson nano.
https://developer.nvidia.com/embedded/jetson-linux-archive

pythops, 2024-06-25:
for jetson nano, try this instead
```
just build-jetson-image -b jetson-nano -r 100 
```

CristianEduardMihai, 2024-06-25:
`just build-jetson-image -b jetson-nano -r 100 -d SD -l 32` succesfully built
will try the command you suggested above too
i had to remove all cuda refferences from the Container files, won't use it anyways