pythops/jetson-image #137: Unable to locate package cuda-toolkit

ijiki16, 2024-08-11:
# Build error on  Jetson  Nano 2gb

command to build image: 

`just build-jetson-rootfs 20.04`
`just build-jetson-image -b jetson-nano-2gb -l 32`

output on second command:
```bash
STEP 1/23: FROM ubuntu:jammy
STEP 2/23: LABEL org.opencontainers.image.authors="Badr @pythops"
--> Using cache 0244c9b7f5cc9a999b2aadc2455bff2a0b2136ff96acbf9678eaddfbb287b689
--> 0244c9b7f5cc
STEP 3/23: ARG BSP=https://developer.download.nvidia.com/embedded/L4T/r32_Release_v7.4/T210/Jetson-210_Linux_R32.7.4_aarch64.tbz2
--> Using cache 53181e60833eb27c76451567dfa28a87471acb286c9cdab0038147f15d4081cd
--> 53181e60833e
STEP 4/23: ARG L4T_PACKAGES
--> Using cache b20c00c64327a113cca1266867f924d74aeb9a7b184d5c819e900a2b73c1717b
--> b20c00c64327
STEP 5/23: RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio
--> Using cache b2b3a0aa4250faad8ff1137a0cfa7c6e33963339bd8a455aba90d8638921b4fa
--> b2b3a0aa4250
STEP 6/23: RUN mkdir /build &&     wget -qO- $BSP | tar -jxpf - -C /build
--> Using cache ad1a81eee4176eba32d324a53d26bad09c5d5cd5e0eaaa54b771e9fc5e791c30
--> ad1a81eee417
STEP 7/23: COPY rootfs/ /build/Linux_for_Tegra/rootfs/
--> Using cache 472fed75575617935bedcd43913dac61ea440f033183d4a3da35a7d98030e005
--> 472fed755756
STEP 8/23: RUN chmod 4755 /build/Linux_for_Tegra/rootfs/usr/bin/sudo
--> Using cache 61cd7c4f9736fadcce8d1a0005cb0630e9e9f27b68e06fb9cd7bb0973b80bcb2
--> 61cd7c4f9736
STEP 9/23: COPY patches /patches
--> Using cache 7b387c77108f8d3787192fb5f4f432950f938b7c803a818afd092fb2d5aa1a84
--> 7b387c77108f
STEP 10/23: RUN patch /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh < /patches/nv-apply-debs.diff &&     patch /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh < /patches/jetson-disk-image-creator.diff
--> Using cache 59211285cccf37456350e247c9392d01a05ef431eab63a816e9d00244714da90
--> 59211285cccf
STEP 11/23: RUN rm -rf /build/Linux_for_Tegra/tools/python-jetson-gpio_*_arm64.deb
--> Using cache 52932cd587f493a0cc0135d91c76e43c8a79c803c2348a76ac70293bb50b7e66
--> 52932cd587f4
STEP 12/23: WORKDIR /build/Linux_for_Tegra/
--> Using cache 6c34b4099b926240d42559377ff479a72cd168f9119d804772d113cd52892cc1
--> 6c34b4099b92
STEP 13/23: RUN ./apply_binaries.sh
--> Using cache 2e3667f32f1e51a573a0a609c9542b61dd057b9b1b6213d8c007077abe180b51
--> 2e3667f32f1e
STEP 14/23: RUN chroot /build/Linux_for_Tegra/rootfs/ /bin/bash -c "chown -R jetson:jetson /home/jetson && chmod 1777 /tmp"
--> Using cache 448dc7fd44c2b84bd0f029a81b952b662ad42ec3c840b27e21ba4958b289bd3c
--> 448dc7fd44c2
STEP 15/23: RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev &&     chroot rootfs /bin/bash -c "echo 'nameserver 8.8.8.8' > /etc/resolv.conf && apt update; apt install -y cuda-toolkit $L4T_PACKAGES"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [128 kB]
Get:3 https://repo.download.nvidia.com/jetson/common r32.7 InRelease [2555 B]
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Ign:6 https://repo.download.nvidia.com/jetson/<SOC> r32.7 InRelease
Hit:7 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Err:8 https://repo.download.nvidia.com/jetson/<SOC> r32.7 Release
  404  Not Found [IP: 2.16.19.26 443]
Get:9 https://repo.download.nvidia.com/jetson/common r32.7/main arm64 Packages [16.6 kB]
Reading package lists...
E: The repository 'https://repo.download.nvidia.com/jetson/<SOC> r32.7 Release' does not have a Release file.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package cuda-toolkit
Error: building at STEP "RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev &&     chroot rootfs /bin/bash -c "echo 'nameserver 8.8.8.8' > /etc/resolv.conf && apt update; apt install -y cuda-toolkit $L4T_PACKAGES"": while running runtime: exit status 100
```

mdbug, 2024-08-11:
I can confirm that. I am having the same problem.

egeakman, 2024-08-11:
Can you try adding the below line to ``Containerfile.image.l4t32`` before setting ``BSP``?

```
ARG SOC=t210
```

ijiki16, 2024-08-12:
@pythops still error, now `WORKDIR`

```bash
STEP 1/22: FROM ubuntu:jammy
STEP 2/22: LABEL org.opencontainers.image.authors="Badr @pythops"
--> Using cache 0244c9b7f5cc9a999b2aadc2455bff2a0b2136ff96acbf9678eaddfbb287b689
--> 0244c9b7f5cc
STEP 3/22: ARG BSP=https://developer.download.nvidia.com/embedded/L4T/r32_Release_v7.4/T210/Jetson-210_Linux_R32.7.4_aarch64.tbz2
--> Using cache 53181e60833eb27c76451567dfa28a87471acb286c9cdab0038147f15d4081cd
--> 53181e60833e
STEP 4/22: ARG L4T_PACKAGES
--> Using cache b20c00c64327a113cca1266867f924d74aeb9a7b184d5c819e900a2b73c1717b
--> b20c00c64327
STEP 5/22: RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio
--> Using cache b2b3a0aa4250faad8ff1137a0cfa7c6e33963339bd8a455aba90d8638921b4fa
--> b2b3a0aa4250
STEP 6/22: RUN mkdir /build &&     wget -qO- $BSP | tar -jxpf - -C /build
--> Using cache ad1a81eee4176eba32d324a53d26bad09c5d5cd5e0eaaa54b771e9fc5e791c30
--> ad1a81eee417
STEP 7/22: COPY rootfs/ /build/Linux_for_Tegra/rootfs/
--> 09abeddf953f
STEP 8/22: RUN chmod 4755 /build/Linux_for_Tegra/rootfs/usr/bin/sudo
--> 038191167c91
STEP 9/22: COPY patches /patches
--> 7482e6c900a6
STEP 10/22: RUN patch /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh < /patches/nv-apply-debs.diff &&     patch /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh < /patches/jetson-disk-image-creator.diff
patching file /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
patching file /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh
--> 53092c179b47
STEP 11/22: RUN rm -rf /build/Linux_for_Tegra/tools/python-jetson-gpio_*_arm64.deb
--> 813ff15f8a82
STEP 12/22: WORKDIR /build/Linux_for_Tegra/
--> d59d26c6f34b
STEP 13/22: RUN ./apply_binaries.sh
Using rootfs directory of: /build/Linux_for_Tegra/rootfs
Installing extlinux.conf into /boot/extlinux in target rootfs
/build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh
Root file system directory is /build/Linux_for_Tegra/rootfs
Copying public debian packages to rootfs
Start L4T BSP package installation
QEMU binary is not available, looking for QEMU from host system
Found /usr/bin/qemu-aarch64-static
Installing QEMU binary in rootfs
/build/Linux_for_Tegra/rootfs /build/Linux_for_Tegra
Installing BSP Debian packages in /build/Linux_for_Tegra/rootfs
Selecting previously unselected package nvidia-l4t-core.
(Reading database ... 48725 files and directories currently installed.)
Preparing to unpack .../nvidia-l4t-core_32.7.4-20230608212426_arm64.deb ...
Pre-installing... skip compatibility checking.
Unpacking nvidia-l4t-core (32.7.4-20230608212426) ...
Setting up nvidia-l4t-core (32.7.4-20230608212426) ...
Selecting previously unselected package jetson-gpio-common.
(Reading database ... 48771 files and directories currently installed.)
Preparing to unpack .../jetson-gpio-common_2.0.17_arm64.deb ...
Unpacking jetson-gpio-common (2.0.17) ...
Selecting previously unselected package python3-jetson-gpio.
Preparing to unpack .../python3-jetson-gpio_2.0.17_arm64.deb ...
Unpacking python3-jetson-gpio (2.0.17) ...
Selecting previously unselected package nvidia-l4t-3d-core.
Preparing to unpack .../nvidia-l4t-3d-core_32.7.4-20230608212426_arm64.deb ...
Unpacking nvidia-l4t-3d-core (32.7.4-20230608212426) ...
Selecting previously unselected package nvidia-l4t-apt-source.
Preparing to unpack .../nvidia-l4t-apt-source_32.7.4-20230608212426_arm64.deb ...
Unpacking nvidia-l4t-apt-source (32.7.4-20230608212426) ...
Selecting previously unselected package nvidia-l4t-camera.
Preparing to unpack .../nvidia-l4t-camera_32.7.4-20230608212426_arm64.deb ...
Unpacking nvidia-l4t-camera (32.7.4-20230608212426) ...
Selecting previously unselected package nvidia-l4t-configs.
Preparing to unpack .../nvidia-l4t-configs_32.7.4-20230608212426_arm64.deb ...
Unpacking nvidia-l4t-configs (32.7.4-20230608212426) ...
[... 92 lines cut ...]
Pre-installing initrd package, skip flashing
Setting up nvidia-l4t-jetson-io (32.7.4-20230608212426) ...
Setting up nvidia-l4t-multimedia (32.7.4-20230608212426) ...
Setting up nvidia-l4t-kernel-dtbs (4.9.337-tegra-32.7.4-20230608212426) ...
Pre-installing kernel-dtbs package, skip flashing
Setting up nvidia-l4t-kernel-headers (4.9.337-tegra-32.7.4-20230608212426) ...
Setting up nvidia-l4t-camera (32.7.4-20230608212426) ...
Setting up nvidia-l4t-gstreamer (32.7.4-20230608212426) ...
Processing triggers for nvidia-l4t-kernel (4.9.337-tegra-32.7.4-20230608212426) ...
Processing triggers for libc-bin (2.31-0ubuntu9.16) ...
/build/Linux_for_Tegra
Removing QEMU binary from rootfs
Removing stashed Debian packages from rootfs
L4T BSP package installation completed!
Rename ubuntu.desktop --> ux-ubuntu.desktop
Disabling NetworkManager-wait-online.service
Disable the ondemand service by changing the runlevels to 'K'
Success!
--> 42d2293bbc6e
STEP 14/22: RUN chroot /build/Linux_for_Tegra/rootfs/ /bin/bash -c "chown -R jetson:jetson /home/jetson && chmod 1777 /tmp"
--> fc08053a5955
STEP 15/22: RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev && WORKDIR /tmp/pythops
/bin/sh: 1: WORKDIR: not found
Error: building at STEP "RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev && WORKDIR /tmp/pythops": while running runtime: exit status 127
```

ijiki16, 2024-08-12:
@pythops it worked, but can't find final image

```bash
STEP 1/23: FROM ubuntu:jammy
STEP 2/23: LABEL org.opencontainers.image.authors="Badr @pythops"
--> Using cache 0244c9b7f5cc9a999b2aadc2455bff2a0b2136ff96acbf9678eaddfbb287b689
--> 0244c9b7f5cc
STEP 3/23: ARG BSP=https://developer.download.nvidia.com/embedded/L4T/r32_Release_v7.4/T210/Jetson-210_Linux_R32.7.4_aarch64.tbz2
--> Using cache 53181e60833eb27c76451567dfa28a87471acb286c9cdab0038147f15d4081cd
--> 53181e60833e
STEP 4/23: ARG L4T_PACKAGES
--> Using cache b20c00c64327a113cca1266867f924d74aeb9a7b184d5c819e900a2b73c1717b
--> b20c00c64327
STEP 5/23: RUN apt update &&     apt install -y         wget         patch         bzip2         qemu-user-static         sudo         libxml2-utils         python3         gdisk         kpartx         cpio
--> Using cache b2b3a0aa4250faad8ff1137a0cfa7c6e33963339bd8a455aba90d8638921b4fa
--> b2b3a0aa4250
STEP 6/23: RUN mkdir /build &&     wget -qO- $BSP | tar -jxpf - -C /build
--> Using cache ad1a81eee4176eba32d324a53d26bad09c5d5cd5e0eaaa54b771e9fc5e791c30
--> ad1a81eee417
STEP 7/23: COPY rootfs/ /build/Linux_for_Tegra/rootfs/
--> Using cache 09abeddf953f560c5e9afc84944de7950856b42228cb65fced97f0229f508fa8
--> 09abeddf953f
STEP 8/23: RUN chmod 4755 /build/Linux_for_Tegra/rootfs/usr/bin/sudo
--> Using cache 038191167c91b8a29de996fb7299ac0eda73a32e40722e7ba721c473cdda2422
--> 038191167c91
STEP 9/23: COPY patches /patches
--> Using cache 7482e6c900a6fad4ac06e3d138f66b312bf6613d3991799fd5cb7d619d7cd0e0
--> 7482e6c900a6
STEP 10/23: RUN patch /build/Linux_for_Tegra/nv_tegra/nv-apply-debs.sh < /patches/nv-apply-debs.diff &&     patch /build/Linux_for_Tegra/tools/jetson-disk-image-creator.sh < /patches/jetson-disk-image-creator.diff
--> Using cache 53092c179b475210199477d86b580709242a723582cf7c3836ef595acea6e594
--> 53092c179b47
STEP 11/23: RUN rm -rf /build/Linux_for_Tegra/tools/python-jetson-gpio_*_arm64.deb
--> Using cache 813ff15f8a822079dfba1ae54810b0ab1f2b7442755509ecd833ee1de019d4df
--> 813ff15f8a82
STEP 12/23: WORKDIR /build/Linux_for_Tegra/
--> Using cache d59d26c6f34badee847c3de13e8efaf66c706051b57176835f6fca5a8bd51ca5
--> d59d26c6f34b
STEP 13/23: RUN ./apply_binaries.sh
--> Using cache 42d2293bbc6e2557719c2cc2f1deb133361e7e1a5f47dbf29ad2a1cde1d50d84
--> 42d2293bbc6e
STEP 14/23: RUN chroot /build/Linux_for_Tegra/rootfs/ /bin/bash -c "chown -R jetson:jetson /home/jetson && chmod 1777 /tmp"
--> Using cache fc08053a595548b483c4a1cae382a3fedb26d309923d8cd01a3ed0dce7a97f07
--> fc08053a5955
STEP 15/23: RUN mount -t proc proc rootfs/proc/ &&     mount -t sysfs sys rootfs/sys &&     mount -o bind /dev rootfs/dev
--> ee3caed322d2
STEP 16/23: WORKDIR /tmp/pythops
--> 200e26782121
STEP 17/23: RUN gzip -d -c /build/Linux_for_Tegra/bootloader/l4t_initrd.img > initrd
--> 5141e9acce96
STEP 18/23: WORKDIR /tmp/pythops/lab
--> 4f883d6572bc
STEP 19/23: RUN cpio -iv < ../initrd &&     echo "nameserver 127.0.0.53" > etc/resolv.conf &&     find . | cpio --create --format="newc" > ../new_initrd
.
proc
tmp
dev
usr
usr/lib
usr/lib/aarch64-linux-gnu
usr/lib/aarch64-linux-gnu/libkrb5support.so.0.1
usr/lib/aarch64-linux-gnu/libk5crypto.so.3
usr/lib/aarch64-linux-gnu/libgssapi_krb5.so.2
usr/lib/aarch64-linux-gnu/libk5crypto.so.3.1
usr/lib/aarch64-linux-gnu/libgssapi_krb5.so.2.2
usr/lib/aarch64-linux-gnu/libkrb5.so.3
usr/lib/aarch64-linux-gnu/libkrb5support.so.0
usr/lib/aarch64-linux-gnu/libkrb5.so.3.3
usr/lib/aarch64-linux-gnu/libcrypto.so.1.1
usr/bin
usr/bin/tail
[... 156 lines cut ...]
# R32 , REVISION: 7.4
###############################################################################
Board ID(3448) version(300) 
copying bctfile(/build/Linux_for_Tegra/bootloader/t210ref/BCT/P3448_A00_lpddr4_204Mhz_P987.cfg)... done.
copying bootloader(/build/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
copying initrd(/build/Linux_for_Tegra/bootloader/l4t_initrd.img)... done.
Making Boot image... done.
Existing sosfile(/build/Linux_for_Tegra/bootloader/nvtboot_recovery.bin) reused.
copying tegraboot(/build/Linux_for_Tegra/bootloader/t210ref/nvtboot.bin)... done.
copying cpu_bootloader(/build/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
copying bpffile(/build/Linux_for_Tegra/bootloader/t210ref/sc7entry-firmware.bin)... done.
copying wb0boot(/build/Linux_for_Tegra/bootloader/t210ref/warmboot.bin)... done.
Existing tosfile(/build/Linux_for_Tegra/bootloader/tos-mon-only.img) reused.
Existing eksfile(/build/Linux_for_Tegra/bootloader/eks.img) reused.
copying dtbfile(/build/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0003-p3542-0000.dtb)... done.
Copying nv_boot_control.conf to rootfs
        populating kernel to rootfs... done.
        populating initrd to rootfs... done.
        populating kernel_tegra210-p3448-0003-p3542-0000.dtb to rootfs... done.
Making system.img... 
losetup: system.img: failed to set up loop device: No such file or directory
mapping system.img to loop device failed.
cp: cannot stat 'jetson.img': No such file or directory
[OK]
```

egeakman, 2024-08-12:
@ijiki16 IIRC I did ``touch system.img`` or ``touch jetson.img`` (I don't remember which one) before running the script. And that solved this issue for me

pythops, 2024-08-13:
This seems to happen only with l4t 32. I have tried to build the image for jetson orin nano with l4t 36 and it works :thinking:

closed 2024-10-05 by pythops

ijiki16, 2024-11-07:
It works for me too on jetson nano 2GB

PR #138 (2024-10-05): Rename cuda_toolkit package for r32
Fixes #137

The problem was in a different package name for `cuda_toolkit` for r32.

I cannot reproduce the second issue mentioned in the ticket - `losetup: system.img: failed to set up loop device`. It is possible that it has been resolved upstream.