pythops/jetson-image #146: A few problems with 24.04 and Jetson Orin Nano

shuhaowu, 2024-12-01:
Thanks for the image! I built a 24.04 image for the Jetson Orin Nano and noticed a few issues:

- It seems like the ethernet device is renamed automatically, which means the netplan file is not automatically causing ethernet to connect with DHCP. The ethernet interface name I'm getting is `enP8p1s0`. I had to adjust the netplan file to make this work:

```
network:
  version: 2
  renderer: networkd
  ethernets:
    eth0:
      match:
        name: en*
      dhcp4: true
    eth1:
      match:
        name: eth*
      dhcp4: true
```

- After install, CUDA programs and nvidia-smi only runs as the root user. This is fixed by adding the `jetson` user to the `render` group: `usermod -a -G render jetson`.

- The time synchronization is not working after boot because `systemd-timesyncd` seems to think the machine is running as a container, due to the `ConditionVirtualizaton=!container` check in the service file. This is a problem as certificate checks rely on the correct time and the system can initialize at Unix epoch. I'm not sure why this is being detected as a container, but I added an override file via `systemctl edit systemd-timesyncd.service` with the following override content:

```
[Unit]
ConditionVirtualization=
```

- The `nv-l4t-bootloader-config.service` seems to not run. This service seems to be for updating the UEFI? It fails with the following logs. I have not figured out how to make it run yet:

```
Jan 01 00:06:33 orinnano systemd[1]: Starting nv-l4t-bootloader-config.service - Configure bootloader service...
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: 3767--0005--1--jetson-orin-nano-devkit-
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1408]: TNSPEC 3767-300-0005-K.2-1-0-jetson-orin-nano-devkit-
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1408]: COMPATIBLE_SPEC 3767--0005--1--jetson-orin-nano-devkit-
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1408]: TEGRA_CHIPID 0x23
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1408]: TEGRA_OTA_BOOT_DEVICE /dev/mtdblock0
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1408]: TEGRA_OTA_GPT_DEVICE /dev/mtdblock0
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: Info: Spec variable TegraPlatformSpec is not found.
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: Info: Write TegraPlatformSpec with 3767-300-0005-K.2-1-0-jetson-orin-nano-devkit-.
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1410]: chattr: No such file or directory while trying to stat /sys/firmware/efi/efivars/TegraPl>
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: Info: Spec variable TegraPlatformCompatSpec is not found.
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: Info: Write TegraPlatformCompatSpec with 3767--0005--1--jetson-orin-nano-devkit-.
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1413]: chattr: No such file or directory while trying to stat /sys/firmware/efi/efivars/TegraPl>
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: Info: The esp is not mounted to /boot/efi.
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1419]: EFI variables are not supported on this system.
Jan 01 00:06:33 orinnano nv-l4t-bootloader-config.sh[1147]: Error: Failed to get BootCurrent by efibootmgr.
Jan 01 00:06:33 orinnano systemd[1]: nv-l4t-bootloader-config.service: Main process exited, code=exited, status=1/FAILURE
Jan 01 00:06:33 orinnano systemd[1]: nv-l4t-bootloader-config.service: Failed with result 'exit-code'.
Jan 01 00:06:33 orinnano systemd[1]: Failed to start nv-l4t-bootloader-config.service - Configure bootloader service.
```

Anyways once again thanks for this image.

(Disclaimer: I'm here on a personal basis and am not affiliated with Jetson or know anything about it)

pythops, 2024-12-01:
Thanks for reporting those issues and the fixes for them. 
I will look into the last point, that one I was not aware of

shuhaowu, 2024-12-06:
Another one that I remembered:

`newgrp` doesn't have the setuid bit on it so it doesn't work. Fixed with chmod +s.

pythops, 2025-01-18:
I don't have much time these days but PRs are always welcome :)