pythops/jetson-image #150: Add support for the  Jetson Orin Nano Super board

pythops, 2024-12-18:


roop90, 2025-01-17:
Hi there,

I own an Orin Nano Super and it works like a normal Orin.
Running with 24.04 and L4T 36.x.
I suffer the same Issues as stated in #146, tho.

Greetings,
roop

roop90, 2025-01-18:
A little update:

As far as I can tell 24.04 isn't working as well as I thought.
It ran for a while and after a reboot services won't bind to local ports for whatever reason. At this point I get an IP and can ping, but DNS and SSH no longer work.
Not sure if this is only related to the Orin Nano Super or a general problem with 24.04.

I will try 22.04 at this point.

TWhidden, 2025-11-21:
I ran into this also - but found the following:

Ubuntu 24 seems to use DNS DNSSEC - and if you don't have a real time clock, and NTP hasn't updated, it won't be able to resolve.  I could have done the allow downgrade but turning of DNSSEC seemed to resolve it for me.  I also had to wait for network to go up before loading `systemd-networkd`.  I am not a linux expert here, but after several build iterations, this got me to success.

This worked for me (I have some custom scripts running before the packaging of the IMG):

```bash
# ============================================================================
# DNS Configuration
# ============================================================================
echo "Configuring systemd-resolved..."

# Remove any existing configs to ensure clean slate
rm -f /etc/systemd/resolved.conf 2>/dev/null || true
rm -rf /etc/systemd/resolved.conf.d 2>/dev/null || true

# Create authoritative configuration
cat > /etc/systemd/resolved.conf << 'EOF'
[Resolve]
DNS=1.1.1.1 1.0.0.1
FallbackDNS=8.8.8.8 8.8.4.4
# Does not work well on RTC-less systems like Jetson Nano/Orin
DNSSEC=no
Cache=no-negative
EOF

ln -sf /run/systemd/resolve/stub-resolv.conf /etc/resolv.conf 2>/dev/null || true

# Configure systemd-networkd to wait for network to be online
mkdir -p /etc/systemd/system/systemd-networkd-wait-online.service.d
cat > /etc/systemd/system/systemd-networkd-wait-online.service.d/wait-any.conf << 'EOF'
[Service]
ExecStart=
ExecStart=/lib/systemd/systemd-networkd-wait-online --any
EOF

echo "✓ systemd-resolved configured (DNSSEC disabled for RTC-less systems)"
```

Hope that helped